technyon/nuki_hub #97: Cannot use let's encrypt certificate

chatziko, 2023-01-27:
I'm having trouble enabling TLS in MQTT and wondering whether it's a bug or I'm missing something. I have configured mosquitto (HA addon) to use a certificate from let's encrypt (__not__ self-signed), and I can connect to it from various places using the following ca file: https://letsencrypt.org/certs/isrgrootx1.pem

Eg
```
mosquitto_pub --cafile isrgrootx1.pem <host> ...
```

But it doesn't work in nuki hub. In fact, I can't even upload this file since the "MQTT SSL CA Certificate" field is 1500 chars max, while the cert (in PEM format) is larger. Should I use some other format?

technyon, 2023-01-27:
@rodriguezst Can you help here?

chatziko, 2023-01-27:
Maybe 1500 bytes is the limit in binary form? (but certainly not in the PEM format). Btw I tried patching the textarea limit in chrome, but I think there is also a 1500 chars limit in the actual code.

technyon, 2023-01-27:
What's a reasonable value? In the end they have to be stored in flash and in memory.

mundschenk-at, 2023-01-27:
Found some thoughts on PEM file size here: https://groups.google.com/g/mailing.openssl.users/c/awvGfcOwfZM

chatziko, 2023-01-27:
There doesn't seem to be a hard limit since keys can in theory be arbitrarily large and certs can also contain metadata. But the larger root cert in my `/etc/ssl/certs` is 2772 bytes.

technyon, 2023-01-28:
What about going with 2048? Recommended in the google groups discussion and it would fit @chatziko 's pem file.

chatziko, 2023-01-28:
2048 would cover 85% of my `/etc/ssl/certs`. 2200 would push it to 96%.

technyon, 2023-01-28:
There are three definitions for certificates. What about TLS_CERT_MAX_SIZE and TLS_KEY_MAX_SIZE ?

add_compile_definitions(TLS_CA_MAX_SIZE=1500)
add_compile_definitions(TLS_CERT_MAX_SIZE=1500)
add_compile_definitions(TLS_KEY_MAX_SIZE=1800)

chatziko, 2023-01-28:
For the key 1800 bytes should be ok for 2048 rsa (of course a large key wouldn't fit, but I doubt anyone will use larger keys for mqtt). For the cert no idea, but the client cert will likely be self-generated, and with the standard way 1500 bytes is ok.

mundschenk-at, 2023-01-28:
I think by default (unless the documentation explicitly tells me otherwise in very bold type), I'd create a 4096 bit key for everything.

chatziko, 2023-01-28:
I doubt you get any practical improvement in security, and the performance penalty for devices like the esp32 would be non-negligible.

mundschenk-at, 2023-01-28:
You are very likely right, I'm just saying it should be prominently documented because I'm not sure that the assumption "nobody would use larger keys" holds.

chatziko, 2023-01-28:
Good point indeed, it took me some time to realize that the cert I was copy/pasting was being truncated. The size limit should be stated in the config page.

chatziko, 2023-01-28:
> I tested the different options (CA only, CA & client validation) with test.mosquitto.org and everything was working fine.

@rodriguezst : Is your connection to test.mosquitto.org stable?

When connecting to test.mosquitto.org port 8883 I see in the log:
```
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connected
Nuki start pairing
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connected
Nuki start pairing
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connected
Nuki start pairing
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connected
Nuki start pairing
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connect failed, rc=-2
ERROR - Generic error
```
Looks like the connection succeeds but then it disconnects a few seconds later.

But when I use port 1883 the connection seems stable:
```
MQTT: Connecting without credentials
MQTT connected
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
```

chatziko, 2023-01-28:
Btw I just noticed that test.mosquitto.org also offers TLS with a let's encrypt cert on port 8886 !

I tried it (with [isrgrootx1.pem](https://letsencrypt.org/certs/isrgrootx1.pem)), after increasing the 1500 limit to 2200, and it works fine! Except from the fact that the unstable behaviour shown above also happens.

technyon, 2023-01-28:
Increasing only TLS_CA_MAX_SIZE is enough? Should be fine, we just cleared ~5k of memory by removing CSS content.

chatziko, 2023-01-28:
I made a build with just `TLS_CA_MAX_SIZE = 2200` and it works fine, tested with both
- test.mosquitto.org:8886 (pretty useful, it uses let's encrypt)
- my own mosquitto

Except from the "instability" issue I just noticed and reported above, which seems a whole different issue anyway.

I think it's worth making a release with `TLS_CA_MAX_SIZE = 2200` at this point.

technyon, 2023-01-28:
Not directly related, but could someone check this binary. Exchanged the mqtt library yet again, because the new also has problems, want to make sure encryption still works. This binary has a ca size of 2200.

[nuki_hub-esp-mqtt-7.zip](https://github.com/technyon/nuki_hub/files/10528055/nuki_hub-esp-mqtt-7.zip)

chatziko, 2023-01-28:
Maybe I did something stupid, I took the latest release, replaced nuki_hub.bin and ran
```
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader_qio_80m.bin 0x10000 nuki_hub.bin 0x8000 nuki_hub.partitions.bin
```

Then I got a crash while testing with test.mosquitto.org:8883
```
WiFi connected: 192.168.3.231
Host name: nukihub
MQTT Broker: test.mosquitto.org:8883
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401c01fe  PS      : 0x00060630  A0      : 0x800d2fe1  A1      : 0x3ffcedb0  
A2      : 0x00000000  A3      : 0x3ffbc0c8  A4      : 0xfc15bd5f  A5      : 0x3ffc62b8  
A6      : 0x00000004  A7      : 0x3f40a67c  A8      : 0x00000001  A9      : 0x3ffced60  
A10     : 0x00000002  A11     : 0x3f409b5c  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000007c  LBEG    : 0x40091321  LEND    : 0x40091331  LCOUNT  : 0xfffffffe  

Backtrace: 0x401c01fb:0x3ffcedb0 0x400d2fde:0x3ffcedd0 0x400d262e:0x3ffcee50 0x40122892:0x3ffceed0

ELF file SHA256: e53606ece04fbae7

E (988) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (457) esp_core_dump_flaKr��core dump partition found!
E (457) esp_core_dump_flash: No core dump partition found!
Using PIN 26 for network device selection
Network device: Builtin WiFi
MQTT without TLS.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: koko-isolatedlan
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.3.231
```

technyon, 2023-01-28:
Does this always happen, or is it a single crash?

chatziko, 2023-01-28:
Right now on every boot I get this, I can't even access the web interface to change the settings.

chatziko, 2023-01-28:
Flashing back v7.3 (and no other change in settings) it boots fine and manages to connect to test.mosquitto.org:8883 (although with the instability reported above).

technyon, 2023-01-28:
How do I connect to test.mosquitto.org:8883 ? I changed broker, port and copy / pasted the certificate, but it can't connect.

chatziko, 2023-01-28:
You need [this cert](https://test.mosquitto.org/ssl/mosquitto.org.crt) and no username/password.

technyon, 2023-01-28:
Doesn't work ... just to make sure this into "MQTT SSL CA Certificate" right?

technyon, 2023-01-28:
My fault I didn't switch to Wifi ...

technyon, 2023-01-28:
I could reproduce the crashed. 8.0 is out, the crash is fixed.

chatziko, 2023-01-29:
Just tested it, the release looks awesome. Connects fine (with let's encrypt certs) and the disconnects are gone. Closing this, thanks @technyon.

closed 2023-01-29 by chatziko