technyon/nuki_hub #101: Esp32 blocked?

testHAtest, 2023-01-30:
Hello everyone, it happens every now and then that the esp32 seems to freeze, the red LED flashes very quickly but it doesn't let you do any lock action and the statuses don't update them, I try to access via browser and I reach it, wifi working and I access all the configurations, the mqq is connected but it seems blocked, I restart and all the states are updated and the lock commands work. What could it be? Thanks so much

technyon, 2023-01-30:
The blinking is usually the logs written to the serial line. Can you try get logs ... either by enabling MQTT logging or by reading them over USB. You can use for example hterm, baud rate is 115200.

technyon, 2023-01-31:
Also please recheck with release 8.1

sebax80, 2023-02-01:
Thanks, I'm downloading it now and updating. further updates coming soon

testHAtest, 2023-02-01:
[image]
:-)

technyon, 2023-02-01:
So I can close the issue?

testHAtest, 2023-02-02:
does it fix with 8.1? I'd like to know if you understood what the problem was.

technyon, 2023-02-02:
There was an issue in the mqtt connection logic which was fixed in 8.1

testHAtest, 2023-02-03:
Perfect and thank you for the fantastic work you do!!!! and yes you can close the issue!

closed 2023-02-03 by technyon