technyon/nuki_hub #102: configuration/autoLock resetting on change

adamlc, 2023-01-31:
Hey, just want to say what a great project this is and its been working pretty flawlessly!

I wanted to integrate the `configuration/autoLock` into my home assistant to not auto lock the door in certain situations.

If I send a `0` MQTT message to `configuration/autoLock` it just changes back to `1` after a few seconds and doesn't change anything. Am I doing something wrong? I've tried via some assistant and MQTT explorer.

The ESP is paired as a bridge rather than an app if that makes any difference :)

Thanks!

technyon, 2023-01-31:
Did you provide your PIN (under "Credentials")?

adamlc, 2023-01-31:
I was missing the PIN, that was it thanks! Now I feel silly :)

closed 2023-01-31 by adamlc

technyon, 2023-01-31:
It's not all that obvious you the PIN. Some functionality needs it, some not.