technyon/nuki_hub #104: Nuki lock opening/closing operation register

amastrogiacomo1968, 2023-01-31:
Staying at home I happened to hear that the Nuki lock opened the door without anyone having given the command, I changed the password immediately but the thing recurs from time to time.
Is there any way to track open/close commands and maybe even which user sent the command?
Thank you

ps Not a real problem (I hope) but more of an implementation request

technyon, 2023-01-31:
You're not at all using the Nuki App? I'd think it's kind of essential to use the lock. I can't tell you very much about tracking in home assistant since I'm not using it. But there are a few HA users around who maybe can help you.

amastrogiacomo1968, 2023-01-31:
Why do you say it is essential? Apart from the initial configuration, I haven't used it anymore but I do everything via Home Assistant, if I really need to go via the web (but first I have to reconnect the Nuki bridge which is now resting in its box).

technyon, 2023-01-31:
There are a lot of settings that can be configured via the app, firmware upgrades, connecting accessories like fobs or a door sensor. If it works for you no worries.

amastrogiacomo1968, 2023-01-31:
I "played" with the configurations at the beginning but it's been a long time now that I haven't changed anything (only the password the other day for the ghost openings of the lock), perhaps the only thing is the updates

mundschenk-at, 2023-01-31:
There is a log you can access via the official app, you should see what user or program caused the unlock event.

amastrogiacomo1968, 2023-01-31:
Thanks, I didn't remember there was the log thanks

alexdelprete, 2023-02-01:
Check also if you created some scripts/automations in HA...if the device is only controlled via HA, it means ONLY some HA action/script/event/automation triggered the command.

amastrogiacomo1968, 2023-02-01:
> Check also if you created some scripts/automations in HA...if the device is only controlled via HA, it means ONLY some HA action/script/event/automation triggered the command.

I just have a script that invokes lock opening no automation; I have automations that block the lock (for example when arming the burglar alarm).
Is there any way to track who and when ran the opening script? Other than putting it in the configuration history? Thank you

mundschenk-at, 2023-02-01:
You should see it in your HA logbook. Also you can see the last few script executions via traces (Automations & Scenes / Scripts and then Traces in the dots menu for the script).

amastrogiacomo1968, 2023-02-01:
> You should see it in your HA logbook. Also you can see the last few script executions via traces (Automations & Scenes / Scripts and then Traces in the dots menu for the script).

I didn't know that in the trace there was also the user id that called the script, this thing is very interesting

technyon, 2023-02-04:
Can this be closed?

amastrogiacomo1968, 2023-02-04:
Yes

closed 2023-02-04 by amastrogiacomo1968