technyon/nuki_hub #105: problem in MQTT configuration

pilot1981, 2023-01-31:
Hi,
I have problem with my MQTT config; I done it:
1) integration
2) add MQTT

itself appeared both entities related to my Nuki smart lock devices...but I'm not able to control them using MQTT commands....but if I use another MQTT client all works fine...

I notice, to use another MQTT client on my HA I addedd these lines in my configuration.yaml:

mqtt:
##SEZIONE SWITCH DI TIPO MQTT
  switch:
## lavatrice
   - name: "Lavatrice"
     command_topic: "cmnd/Lavatrice/power"
     state_topic: "stat/Lavatrice/POWER"
     availability_topic: "tele/Lavatrice/LWT"
     payload_available: "Online"
     payload_not_available: "Offline"
     payload_on: "ON"
     payload_off: "OFF"
     retain: false
##SEZIONE SENSOR DI TIPO MQTT
  sensor:
## misurazione corrente presa WIFI Lavatrice
   - name: "WATT Lavatrice"
     state_topic: "tele/Lavatrice/SENSOR"
     value_template: "{{ value_json['ENERGY'].Power }}"
     unit_of_measurement: "W"
     icon: mdi:gauge

Do you suggest to me to add also these lines in my configuration.yaml? you already know which lines I have to add?

mqtt:
##SEZIONE SWITCH DI TIPO MQTT
  switch:
## lavatrice
   - name: "Nuki1"
     command_topic: "cmnd/nuki1/lock?????"
     state_topic: "stat/nuki1????"
     availability_topic: "tele/nuki1??????"
     payload_available: "Online"
     payload_not_available: "Offline"
     payload_on: "ON"
     payload_off: "OFF"
     retain: false

let me know

regards

technyon, 2023-01-31:
Hi,

Since this is home assistant related issue, I think you're better off asking in the home assistant forum:

https://community.home-assistant.io/

SergioWarriorPeru, 2023-01-31:
> Hi, I have problem with my MQTT config; I done it:
>
> 1. integration
> 2. add MQTT
>
> itself appeared both entities related to my Nuki smart lock devices...but I'm not able to control them using MQTT [...]

Hi,

I use Mosquito Broker and my Nuki appear as a device 

regards

mundschenk-at, 2023-01-31:
Hi @SergioWarriorPeru, the `mqtt` is now configured within the Home Assistant UI and Nuki Hub has auto-discovery out of the box if you set the topic in its MQTT Configuration page. I am assuming what you posted are YAML fragments, but they have got nothing to do with Nuki Hub as far as I can see. What exactly is the question you have?

SergioWarriorPeru, 2023-01-31:
> Hi @pilot1981, the `mqtt` is now configured within the Home Assistant UI and Nuki Hub has auto-discovery out of the box if you set the topic in its MQTT Configuration page. I am assuming what you posted are YAML fragments, but they have got nothing to do with Nuki Hub as far as I can see. What exactly is the question you have?
> 
> Sorry @SergioWarriorPeru, I confused you with the OP.

Not problem...I tried to say that my HA work with Mosquitom integration and Nuki hub without any problem.

pilot1981, 2023-02-01:
I confirm MQTT auto-discover worked fine and MQTT object seems look good...so I go to find my MQTT problems in other place...

pilot1981, 2023-02-02:
very strange...If I launch MQTT command from HA nothing happen...If I use an external MQTT client all works fine and then all works fine also on HA...

after some time again MQTT commands sent from HA don't works...

I don't understand...

SergioWarriorPeru, 2023-02-02:
> very strange...If I launch MQTT command from HA nothing happen...If I use an external MQTT client all works fine and then all works fine also on HA...
> 
> after some time again MQTT commands sent from HA don't works...
> 
> I don't understand...

Hi, I use Node Red in HA
[image]

pilot1981, 2023-02-05:
Hi, 
I done some other tests and I found MQTT command works fine with NUKI smart lock 3.0 but it don’t works with NUKI smart lock 3.0 PRO.

I use 2 differente ESP32 with the same firmware version (the last).

Any idea?

this firmware works fine also with NUKI smart lock 3.0 PRO?

mundschenk-at, 2023-02-05:
@pilot1981 What exactly do you mean by "MQTT commands", BTW, with auto-discovery there are native HA services for almost everything (except of course the lock configuration options). To unlatch the lock, just use the `lock.open` service (whereas to just unlock it, you would use `lock.unlock`).

technyon, 2023-02-05:
The bluetooth API of the Pro lock is no different from the regular lock. It is fully supported and confirmed working with NUKI Hub.

pilot1981, 2023-02-05:
> @pilot1981 What exactly do you mean by "MQTT commands", BTW, with auto-discovery there are native HA services for almost everything (except of course the lock configuration options). To unlatch the lock, just use the `lock.open` service (whereas to just unlock it, you would use `lock.unlock`).

Lock.Unlock and lock.lock doesn’t work

pilot1981, 2023-02-05:
To troubleshoot I would know know to use MQTT logging

mundschenk-at, 2023-02-05:
> To troubleshoot I would know know to use MQTT logging

Enable MQTT logging in the preferences and have something (e.g. a HA automation) listen to the topic `<nuki>/maintenance/log`. The prefix of course depends on what you set in the Nuki Hub configuration. You did set two different prefixes for your two logs?

pilot1981, 2023-02-06:
MQTT auto-discovery don’t works for ti?

mundschenk-at, 2023-02-06:
MQTT logging is not mapped to any HA entities, thus no auto-discovery. You don't need HA for MQTT logging, it is just what I use.

pilot1981, 2023-02-06:
[image]

pilot1981, 2023-02-06:
[image]

mundschenk-at, 2023-02-06:
@pilot1981 Please try to re-pair after unpairing via the `Credentials` page.

mundschenk-at, 2023-02-06:
Can happen. You should increase the number of retries.

pilot1981, 2023-02-06:
I have to increase “ Number of retries if command failed”?

which value do you suggest? 10?

technyon, 2023-02-06:
Maybe go for something like 3 :)

pilot1981, 2023-02-06:
I setup 3 retries, but it don’t works

[image]

pilot1981, 2023-02-07:
I have MQTT connected, Nuki lock paired and now I have also lock state updated correctly, but I ahve always have command failed on Nuki and I found many error in reading config & reading advanced config...

There is a way to collect more data/logs?

mundschenk-at, 2023-02-07:
What's your BLE RSSI for the lock?

pilot1981, 2023-02-08:
[image]
[image]

mundschenk-at, 2023-02-08:
That's pretty low for the lock. Can you try to move the ESP closer?

pilot1981, 2023-02-09:
Hi,
I moved it closer but on HA I always see -74 dbm…very strange…

There is a way to ask an updated value of BLE signal strength?

mundschenk-at, 2023-02-09:
When you view the sensor log, is it a straight line at -74 dBm? Then it's not really updating at all. `RSSI Publish interval` is not set to `-1` on the Nuki Hub?

pilot1981, 2023-02-09:
You are right: RSSI update was disabled.

after I enabled it and moved ESP32 I have -64dbm, but same error in log and no any chance to unlock/lock the door…

I can do other test? I can collect more fat/log/info?

technyon, 2023-02-10:
There must be something wrong with your ESP. Try:

- factory reset your ESP
- try different ESP

pilot1981, 2023-02-12:
Hi,
I tried to flash again original ESP32 mini v3 and it worked again for some hour, but it stop to works again…

So I flash another ESP32 mini v1 but I obtain always same error in MQTT Logs…

technyon, 2023-02-12:
Did you configure the PIN on NUKI Hub?

pilot1981, 2023-02-12:
No, I don’t configured PIN.

yesterday, when I flashed again ESP32 mini v3, I don’t configured PIN and it worked fine soon…

so, I don’t configured PIN in ESP32 mini v1

if I have problem of PIN, I should see error in MQTT logs?

pilot1981, 2023-02-12:
Now I tried to change PIN on NUKI app and update it in ESP32 nukihub, but I obtain again same error in MQTT logs…

pilot1981, 2023-02-25:
I buyed 3 newest ESP32. Tomorrow try to flash it again

technyon, 2023-03-10:
@pilot1981 Did this solve your issue?

pilot1981, 2023-03-10:
I have to wait some days to confirm...we can merge with issue #141 ?

technyon, 2023-03-26:
Closing here let's continue in issue #141

closed 2023-03-26 by technyon