technyon/nuki_hub #111: nuki hub can't handle long password

Eyal51, 2023-02-04:
Using Hass.io with Mosquitto MQTT, can't get it to connect to the broker though it should be the easy part.
When editing "MQTT and Network Configuration" I noticed that the GET request contains the cleartext password, but only the first 30 characters. I'm using the random default MQTT password which is 64 chars long. Tried bypassing it with the correct GET request but no success.

technyon, 2023-02-04:
Since the firmware is running on a microcontroller there are certain limits, and I had to pick a maximum length for a password, which is now 30 characters. Yes, it should be made more obvious in the UI, but you'll have to use a password that's 30 characters or less. I don't think there'll be a big gain in security of using 64 over 30 characters.

bcutter, 2023-02-05:
Just leave a note on the configuration page ("...limited to a maximum of 30 characters"), done.

technyon, 2023-02-05:
It's already done will be in the next release.

closed 2023-02-05 by technyon