technyon/nuki_hub #112: Nuki Keypad 2.0 support?

NdR91, 2023-02-05:
First of all, thank you for this great work @technyon.

Do you have on road map the Nuki Keypad 2 support? 

Thank you!

technyon, 2023-02-05:
No, because there's no documentation available.

See here:
https://developer.nuki.io/t/keypad-2-not-setting-has-keypad-flag-in-config-of-nuki-lock/19539/3

Noone answering unfortunately.

NdR91, 2023-02-05:
Thank you for the feedback. 

Hope to someone answer then!

Thank you again!

closed 2023-02-05 by technyon

mundschenk-at, 2023-02-11:
@technyon They have now updated the documentation it seems.

Eyal51, 2023-02-13:
@technyon I appreciate the work you're doing

technyon, 2023-02-27:
I will look into it as soon as I receive my 2.0 keypad.

bcutter, 2023-02-27:
Let me/us other KP 2.0 users know once I/we can test something 👍