technyon/nuki_hub #113: No MQTT connection ("Client <unknown> disconnected due to protocol error.")

bcutter, 2023-02-05:
NUKI Hub can't connect to HA MQTT broker.

Everything configured correctly, using the same data on another client (e. g. MQTT Explorer on a Windows machine) connects just fine.

Mosquitto broker logs for the NUKI hub:
```
2023-02-05 18:45:03: New connection from 192.168.1.15:52648 on port 1883.
2023-02-05 18:45:03: Client <unknown> disconnected due to protocol error.
2023-02-05 18:45:08: New connection from 192.168.1.15:52649 on port 1883.
2023-02-05 18:45:08: Client <unknown> disconnected due to protocol error.
```

[image]

This is so annoying as I'm trying to make NUKI Hub run for weeks now, testing it every few days. Today I must conclude: the bug is on the NUKI hub side.

Please assist.

mundschenk-at, 2023-02-05:
It looks like you haven't configured MQTT fully on the Nuki Hub side, because otherwise Mosquitto would log your client ID.

bcutter, 2023-02-05:
> It looks like you haven't configured MQTT fully on the Nuki Hub side, because otherwise Mosquitto would log your client ID.

As I wrote: checked that plenty of times, all data is correct. Using the same information from another MQTT client works just fine.

So: tell me what's missing according to your opinion.

[image]

technyon, 2023-02-05:
Please update to 8.4, open the web portal and open the system information page by clicking on the firmware version. Copy & paste your info here.

bcutter, 2023-02-05:
```
Firmware version: 8.4
run: true
deviceId: 3***80***5
mqttbroker: 192.168.1.10
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: nukiopener
maxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
nwhw: 2
nwhwdt: 26
rssipb: -1
hostname: Nuki-Hub
nettmout: -1
restdisc: false
resttmr: -1
rstbcn: -1
lockStInterval: 1800
configInterval: 
batInterval: 1800
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: 43
macb1: 120
macb2: 64
MQTT connected: No
Lock paired: No
Lock PIN set: -
Heap: 80148
Restart reason FW: NetworkDeviceCriticalFailure
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

Note I masked `deviceId` and `hostname` (real one contains an `ü`, just in case that makes any difference).

technyon, 2023-02-05:
No need to censor the deviceId. It's the id of the client (the ESP in this case) and randomly generated when you boot it the first time.

bcutter, 2023-02-05:
Okay. Anything helpful in that log data (which is a great new feature by the way 👍)?

By the way, this is sent when saving the MQTT/network settings:

`http://nuki-hub/method=get?HOSTNAME=Nuki-Hub&MQTTSERVER=192.168.1.10&MQTTPORT=1883&MQTTUSER=username_mqtt&MQTTPASS=[MASKED]&HASSDISCOVERY=homeassistant&MQTTCA=&MQTTCRT=&MQTTKEY=&NWHW=2&NWHWDT=26&RSSI=-1&NETTIMEOUT=-1&RSTDISC=0&RSTTMR=-1&MQTTLOG=0&submit=Save`

For the password, special characters are replaced by some kind of ASCII representation I guess, e. g. the `ü` from the hostname is shown as `%FC`.

Not sure what else I could provide.

mundschenk-at, 2023-02-05:
> Note I masked `deviceId` and `hostname` (real one contains an `ü`, just in case that makes any difference).

That's not a valid DNS label (and I'm pretty sure, ESP32 doesn't translate to punycode).

bcutter, 2023-02-05:
> > Note I masked `deviceId` and `hostname` (real one contains an `ü`, just in case that makes any difference).
> 
> That's not a valid DNS label (and I'm pretty sure, the ESP32 network stack doesn't translate to punycode).

BAM - here we go:

replaced the `ü` in the hostname setting with `ue` and immediately:
[image]

MQTT Explorer:
[image]

Please add input validation or at least a note what exactly is allowed for which fields (same as with the passwort length #111). That would have saved me roughly 3 weeks...

technyon, 2023-02-05:
Max. password length is already added. Input validation at this point would mean the ESP just wouldn't save such a hostname ... which is probably better than nothing since it's more obvious than a failing MQTT connection.

closed 2023-02-05 by technyon