technyon/nuki_hub #114: Wrong default settings in "Advanced NUKI Configuration"

bcutter, 2023-02-05:
_(Everything being said at https://community.home-assistant.io/t/nuki-hub-firmware-for-the-esp32/417572/218)_

Nuki Hub set up with firmware 8.4 gives this as default:

[image]

Result: spamming like crazy (bottom one):
[image]

Messages seem to come down to a normal/comparable amount once the settings have been adjusted.

No idea when this was introduced, somewhere between 7.2 and 8.4, possibly even in latest 8.4?

mundschenk-at, 2023-02-05:
@technyon, I can confirm this for a newly flashed & erased device via the Web Installer:
[image]

I am assuming this has been introduced with the refactoring of the web interface in 8.4.

technyon, 2023-02-05:
Accidentally moved some code instead of copying it. Fixed in 8.5.

closed 2023-02-06 by technyon