technyon/nuki_hub #115: Wifi Latency with Nuki Hub

mkoppanen, 2023-02-06:
I've installed Nuki Hub 8.5 and I am experiencing frequent wifi disconnections and high latency. I've tried a few boards now to rule out a bad board but they all seem to exhibit the same behaviour. In contrast I am running ESPresence on identical board and it seems to have stable connection.

This is a long running ping from nuki hub:

```
--- 192.168.68.130 ping statistics ---
551 packets transmitted, 488 packets received, 11.4% packet loss
round-trip min/avg/max/stddev = 9.783/1262.954/6108.297/1010.591 ms
```

And espresence in the same room connected to same access point:

```
--- 192.168.68.123 ping statistics ---
1402 packets transmitted, 1371 packets received, 2.2% packet loss
round-trip min/avg/max/stddev = 6.381/89.848/990.608/123.983 ms
```

As you can see for some reason average latency on the nuki is 1262.954ms in comparison to 89.848ms on ESPresence. 

Any hints on where to start debugging the issue and is this expected behaviour? My initial thoughts were that could this possibly be related to the fact that ESP32 only has single radio for Bluetooth and Wifi and one of them is hogging it?

Below are logs for the nuki hub (but they don't seem to indicate anything odd):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (496) esp_core_dump_flaKr��core dump partition found!
E (496) esp_core_dump_flash: No core dump partition found!
NUKI Hub version 8.5
Hardware detect     : 1
Hardware detect GPIO: 26
W5500 hardware is disabled, using Wifi.
Network device: Builtin WiFi
MQTT without TLS.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: XXXXXXXX
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.68.130
WiFi connected: 192.168.68.130
Host name: nukihub
MQTT Broker: homeassistant.local:1883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: XXXXXX
MQTT connected
Nuki start pairing
Nuki paired
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Reading config. Result: success
Reading advanced config. Result: success
HASS setup for lock completed.
```

technyon, 2023-02-06:
You're a good candidate to test this special firmware with modified wifi parameters.

[nuki_hub-8.5-sdk-1.zip](https://github.com/technyon/nuki_hub/files/10667362/nuki_hub-8.5-sdk-1.zip)

gatd4, 2023-02-06:
Im getting this with8.5 and sdk version in Home Assistant

The unit of sensor.bedroom_door_lock_wifi_signal_strength (dBm) can not be converted to the unit of previously compiled statistics (dB). Generation of long term statistics will be suppressed unless the unit changes back to dB or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.bedroom_door_lock_bluetooth_signal_strength (dBm) can not be converted to the unit of previously compiled statistics (dB). Generation of long term statistics will be suppressed unless the unit changes back to dB or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this

technyon, 2023-02-06:
Your issue is unrelated and not an actual issue. We changed the unit to dBm because this the correct unit. I'd suggest to clear your historical data.

gatd4, 2023-02-06:
sorry for that

mkoppanen, 2023-02-07:
@technyon I installed the binary linked and having mixed results. Overall the packet loss and latency seems slightly better but disconnections are still there. I have RSSI reporting at 30 second interval and ping monitor from home assistant to the esp32 at 30 second interval. The bluetooth signal graph looks especially odd (the esp32 is about 1 meter away from the lock with a clear line of sight.)

The graphs look like this:

[image]

[image]

mkoppanen, 2023-02-07:
Spoke too soon. Looks like there is still lag but it seems to come at somewhat random intervals after restart

mkoppanen, 2023-02-07:
It looks like the problem appears with nuki hub when connecting to tp-link deco mesh wifi. I set up an old linksys access point with a separate SSID and have not experienced disconnections.

technyon, 2023-02-07:
It's really hard to tell what's going on there. Even if I could reproduce the behaviour, the issue is most likely in the Wifi stack, and that's maintained by Espressif. Why ESPresence works better is hard to tell, they're using the same stack (slightly older even)

mkoppanen, 2023-02-08:
Interesting enough I installed nuki hub on M5Stack Atom Lite and things have been stable ever since.

technyon, 2023-02-08:
That's strange, maybe the old ESP had some kind of fluke?

closed 2023-02-24 by technyon