technyon/nuki_hub #117: Keypad failed attempt information

deadly667, 2023-02-07:
Hi, is there maybe a way how to get information about failed attempt on Keypad. So that we know when someone entered wrong passcode (tried to guess it). With Nuki Bridge there is possibility to set up mail notification in case of failed attempt but to be honest I never succeed to setup it up. 

If there is a way I can help with testing/troubleshooting so that we add that as feature.

technyon, 2023-02-07:
You could extract this information from the "log" that is published when you enable "Publish auth data". You'll get the last 5 log entries (same as in the smartphone app). If the latest one has a completionStats of "invalidCode", you know someone tried to enter a code.

sli-cka, 2023-03-19:
If you are a Home Assistant user you can user, the following [MQTT Sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) to your configuration.yaml:

```
mqtt:
  sensor:
    - state_topic: "nuki/lock/log"
      name: "Nuki Lock Keypad Status"
      value_template: >-
        {% for state in value_json %}
          {% if state.type == "KeypadAction" %}
            {{ state.completionStatus }}
          {% endif %}
        {% endfor %}`
      icon: mdi:dialpad
```
(Maybe you need to adjust the `state_topic`)

This will create a new entity named `sensor.nuki_lock_keypad_status` and display the Status of the last Keypad Action (if there is a matching entry in the last 5 log items). Examples for the Status are `success` or `invalidCode`.

Then you can use this entity to create an automation that notifies you whenever the status changes to `invalidCode`

technyon, 2023-03-19:
Very nice I think we can add this as an auto discovery topic.

sli-cka, 2023-03-19:
That would be even better! :)

deadly667, 2023-03-21:
Thanks both of you. I will close this issue as we solved it. 

For the future reader.

If your nuki/lock/log is just "--" and you have marked "Publish auth data" box that means that you didn't provide correct PIN to Nuki Hub. 

Also Home Assistant automation looks like this after you add @sli-cka part of the code or in that time auto discovery will be set up already: 

`alias: "Hall: Keypad invalid code notification"
description: ""
trigger:
  - platform: template
    value_template: "{{ 'invalidCode' in states('sensor.nuki_lock_keypad_status') }}"
condition: []
action:
  - service: notify.telegrambot
    data:
      message: "🚨 Keypad invalid code entered! 🚨 "
mode: single`

technyon, 2023-03-24:
Check release 8.19