technyon/nuki_hub #118: Unable to update via OTA

amastrogiacomo1968, 2023-02-07:
Since yesterday I have this message when I try to update via OTA "OTA functionality not ready. Please wait a moment and reload.". I'm stuck at 8.3 How can I fix it without connecting the ESP32 to the pc? Thank you

technyon, 2023-02-07:
Do as the instruction says :). After booting the Esp wait 60 seconds, then reload. The OTA function is locked for 60 seconds after booting the ESP.

amastrogiacomo1968, 2023-02-08:
The problem appeared using 2 different laptops, edge and Chrome as a browser and after restarting the esp32 several times and waiting much longer than the indicated 60 seconds. On the mqtt side, no problem, the lock responded to commands quickly. After the flash the problem disappeared and hopefully it won't recur (I'm going to hide the hub inside an empty electrical box near the door). Thanks for the tips

technyon, 2023-02-08:
ok, I'll close the issue for now, we can reopen it if the problem appears again. If so, please copy & paste the details of the system information page here (click the firmware version).

closed 2023-02-08 by technyon