technyon/nuki_hub #119: Hub only works when connected to the laptop

thor9898, 2023-02-08:
I'm having an issue when I try to power the different esp32's to phone chargers. I've tried with 5V 2A, 5V 3A and 5.25V 1A(the one included with the firestick). When I try and power them up with these chargers, the boards keep rebooting and don't connect to the lock, but when I power them up using the USB of my laptop (located in the same place as the phone charger) the boards don't reboot as often and the connection to the lock is very stable and fast.
Here is the difference in uptime when powered with the laptop and with the phone charger
[image]

Here are my configs, which i didn't change when I changed the power supply
[image]
[image]

technyon, 2023-02-08:
That's a very unique issue you have. Some points:

- Which NUKI Hub version are you on?
- Can you post details the system information page? (Click on the firmware version; only available in recent versions)
- Are you using the same USB cable for connecting to the laptop and to the power supply. If possible, test a different cable.
- Do you have another ESP to try?

I'd especially check the last two points, as odd as it is it sounds like a hardware issue.

thor9898, 2023-02-08:
- I have noticed this issue since 8.5, as I tried to make it work since I discovered this in 8.3 version and was having this issues, and discovered how stable it was with the laptop when trying to see if the logs would give me any clue of what my problem was, but ended up with no problem at all when connected to the laptop. I'm now on 8.7
- The cable I think was always the same, ill try change the cable and let you know if I notice any problem.
- I've tried 3 different esp32 boards and I'm having the same issue with the 3 of them.

This is my system info:

Firmware version: 8.7
run: true
deviceId: 1010125340
mqttbroker: 192.168.1.198
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
nwhw: 2
nwhwdt: 26
rssipb: -1
hostname: nukihub
nettmout: 5
restdisc: true
resttmr: -1
rstbcn: 10
lockStInterval: 10
configInterval: 5
batInterval: 30
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: 30
hasmac: true
macb0: -97
macb1: 53
macb2: 10
MQTT connected: Yes
Lock paired: Yes
Lock PIN set: No
Network device: Built-in Wifi
Heap: 80376
Restart reason FW: NetworkDeviceCriticalFailure
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart. 

Thanks for assisting me with this issue <3

technyon, 2023-02-08:
Please go to the mqtt and network configuration and change the network device to Disable W5500. After that with some luck you can track the restart reason either in the sysinfo page (last two lines), or MQTT under maintenace.

thor9898, 2023-02-09:
I just changed it, ill let you know how it goes, although since last night it started working flawless all of a sudden 😅😅😅
If it keeps working i´ll close this issue.
Thanks for the help

closed 2023-02-10 by thor9898