technyon/nuki_hub #120: GPIO to unlatch smart lock not working

wolled, 2023-02-09:
hello, I have successfully paired the nice nuki bridge with my Nuki SmartLock 2.0.
When sending commands to the smartlock it works just fine (lock/unlock/unlatch), but when I am connecting GPIO (GND - Pin27) to unlatch the smarklock the ESP reboots every second time.  Once in a while the Smartlock opens, somtime nothing happens. After a few tries the nuki bridge (ESP) is unresponsive over WLAN and is also not firing the "unlatch" command when PINs are connected. Is this a known issue? If further infos are needed, klindly let me know and I am happy to provide.

technyon, 2023-02-10:
Fixed, will be in the next release.

wolled, 2023-02-10:
wow.  very impressive response. I am looking forward testing it once released and revert back wiht my results. thanks so much 
for now

technyon, 2023-02-10:
You can test this prerelease already.

[nuki_hub-8.8-pre-2.zip](https://github.com/technyon/nuki_hub/files/10710389/nuki_hub-8.8-pre-2.zip)

wolled, 2023-02-11:
Hey, it‘s awsome, it works incredibly well. Thanks so much for your fix, the coffee is well deserved 😊

closed 2023-02-18 by technyon