technyon/nuki_hub #121: Battery level occasionally dropping to 0

chatziko, 2023-02-10:
Hello,

I have connected Nuki to Home Assistant via the MQTT integration, with the default settings (it was discovered automatically). It generally works fine, with a small issue: the reported battery level occasionally jumpts to either 0% or 126% (normally it is 94%). See the plot below.

[image]

Every time this happens, it is restored after exactly 10 minutes (I guess this is the update interval).
Not a big issue obviously, but it messes with some "low battery" automations.

technyon, 2023-02-10:
Hi,

this most likely happens because the battery state is published, but querying from the lock was not successful. Will be addressed in the next release.

chatziko, 2023-02-11:
Awsome, thanks for the quick response. Just updated to 8.8, let's see what happens...

mundschenk-at, 2023-02-11:
I've seen this about two or three times per week, so we will have to monitor for a few days to ascertain the issue is definitely fixed.

bcutter, 2023-02-12:
Exactly what I see and was wondering about too. Different lock types with different battery types.

[image]

The "return from zero" happens differently, for one lock after exactly 1 hour (which is my set battery state query interval) the other only few minutes.

technyon, 2023-02-12:
@bcutter which firmware version is this?

bcutter, 2023-02-12:
It changed during that time. Mainly 8.5 (updated now to 8.8).

technyon, 2023-02-12:
It's one of the points that should be fixed in 8.8.

chatziko, 2023-02-13:
2 days without a single drop since upgrading to 8.8, so this looks like fixed, many thanks!
Closing the issue, I'll reopen if it happens again.

closed 2023-02-13 by chatziko