technyon/nuki_hub #122: Restart through http/mqtt command

bartsnijder, 2023-02-10:
My bluetooth link between hub and lock is sometimes broken which makes the lock uncontrollable through the hub. I recently have set the "Restart if bluetooth beacons not received" option so I should be safe from now on (to double check). However, since the lock opens the main entry I would like to have a second fallback. Is there a http/mqtt command to restart the Nuki hub manually. A broken bluetooth link is detectable through MQTT (eg commandresult = error), so I would like to be able to restart the hub manually when such situation would occur.
Thanks!

technyon, 2023-02-10:
Set the "maintenance/reset" node to 1 to restart the ESP.

bartsnijder, 2023-02-10:
That easy, thanks!

closed 2023-02-10 by bartsnijder