technyon/nuki_hub #126: Lost connection: no control, no state sync, no reliability

bcutter, 2023-02-14:
### I can't control my smart locks using HA anymore:
- Lock/Unlock commands are not performed anymore by the locks
- Status updates from the locks (locking/unlocking locally using button) are not visible in HA

### More information:
- Both Nuki hubs run firmware 8.8
- Both Nuki hubs are reachable via WiFi, constant and stable (not a single echo reply missing)
- Webinterface ob both Nuki hubs is working fine
- MQTT topics have payloads (everything looking good from that point of view, e. g. `/lock/action` etc.)
- Nuki Hub uptimes:
  [image]
- Log Nuki hub 1: see *1
- Log Nuki hub 2: see *2

### Actions:
- I reloaded the MQTT integration in HA: no impact
- I restarted the Mosquitto broker in HA: no impact

### Possibly related / similar to:
- https://github.com/technyon/nuki_hub/issues/51 ?
- https://github.com/technyon/nuki_hub/issues/28 ?

### Workaround (still spooky):
1. Initially I wanted to restart both Nuki hubs (webinterface, `/nukicfg?`, hit `Save` as there's no _dedicated "restart Nuki hub" button yet or - FEATURE REQUEST - can't we allow to restart Nuki hub using MQTT so I can create a restart button/script in HA for emergency cases?_)
2. Instead I just "Enable MQTT logging" first for the Nuki hub 1/door A.
3. After that, I used HA to send a lock.lock command to the lock. The lock locked fine - but the state in Nuki hub and HA still did not update (still "unlocked")!
[image]
4. I disabled MQTT logging again, after a few seconds the state switched to "locked" (also in HA):
[image]
5. Sending a lock.unlock command from HA still did not work (again).
6. At that point I decided to physically power-cycle the Nuki hub 1 for door A. **Frustration overload!** That still did not help: after an initial state sync (if changed), there's no control possible anymore.
7. Note: performing step 1 on the second Nuki hub for door B worked just fine. After that, status sync and control were back fully operational.

### Summary:
In the end, this integration was useless for two daily use-cases today. I had to use the Nuki app on the smartphone to access my home instead of relying on Nuki hub in Home Assistant. That's of course a show-stopper which is why I really want to debug and solve this. A bit frustrating this (Nuki hub) silently stops working and leaves you without control when you rely on it.

### Questions:
1) What could be the reason for this? Felt like the bluetooth component of Nuki hub "died".
2) Any other debug information needed for suches cases?
3) Any settings I should use to work around this issue? Maybe `Enable MQTT logging`?
4) **MOST IMPORTANT ONE: How can I reconnect the Nuki hub 1 to door A (make it functional again)? They seem to not interact anymore...**

----

*1
```
Firmware version: 8.8
run: true
deviceId: 3397801125
mqttbroker: xxx.xxx.xxx.xxx
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki_door_a
openerena: false
mqttoppath: nukiopener
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
nwhw: 2
nwhwdt: 26
rssipb: -1
hostname: ESP-Nuki-Hub-Door-A
nettmout: -1
restdisc: true
resttmr: -1
rstbcn: -1
lockStInterval: 3600
configInterval: 0
batInterval: 7200
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: 43
macb1: 120
macb2: 64
MQTT connected: Yes
Lock paired: Yes
Lock PIN set: Yes
Network device: Built-in Wifi
Heap: 77260
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

*2
```
Firmware version: 8.8
run: true
deviceId: 3045297052
mqttbroker: xxx.xxx.xxx.xxx
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki_door_b
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
nwhw: 2
nwhwdt: 26
rssipb: -1
hostname: ESP-Nuki-Hub-Door-B
nettmout: -1
restdisc: true
resttmr: -1
rstbcn: -1
lockStInterval: 3600
configInterval: 0
batInterval: 7200
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: -41
macb1: 80
macb2: 112
MQTT connected: Yes
Lock paired: Yes
Lock PIN set: Yes
Network device: Built-in Wifi
Heap: 77804
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

bcutter, 2023-02-14:
Updated information on unpairing and pairing. Did not solve the problem, Nuki hub 1 still not functional with smart lock door A. Help appreciated, running out of ideas (and time) here.

mundschenk-at, 2023-02-14:
Have you enabled `Restart if bluetooth beacons not received (seconds; -1 to disable)` in `Advanced Nuki configuration`? I'd set it to 60 seconds. When you connect using WiFi, sometimes the BLE stack crashes and can only be revived by restarting the ESP. That watchdog accomplishes that.

bcutter, 2023-02-14:
Funnily that's what I discovered right now. Set it to 60 for both Nuki hubs:
- Nuki hub 1: set, but still not operational
- Nuki hub 2: once saved and restart initialized, the ESP32 went offline (WiFi) and took a few minutes (!) to come back online.

I did not have the RSSI values reported yet.

Nuki hub 1 still not operational.

bcutter, 2023-02-14:
Bluetooth RSSI is just fine for the Nuki hub 1 (interval set to 60 seconds since few minutes):
[image]

After every Nuki hub restart there's a short time status or commands are sent. After a quite short time this stops.

Any ideas left?

bcutter, 2023-02-14:
Reflashed the still out-of-service Nuki hub with 8.8 - no change.

mahikeulbody, 2023-02-14:
Try to erase fully your ESP32 and then to re-flash it through usb cable. I did not have a so bad situation as your but my nuki hub uptime average was 200' and I had 2,5% of battery consumption per day. After doing that (and keeping the default settings), I got very long uptimes and 2% of battery consumption per 10 days. I have no idea to explain that but since there were no other changes...

technyon, 2023-02-15:
You could also try factory-resetting the lock ... at least that's how I got my opener to work again. Also as already mentioned, completely erasing before flashing can help too.

MQTT logging can help as long as the problem is  on bluetooth side and the ESP stays on Wifi. 

There's an option to trigger a restart via mqtt. Check lock/maintenance/reset.

I have a feeling the bluetooth watchdog should be enabled by default.

bcutter, 2023-02-15:
1. I fully erased the ESP32 and flashed it with Nuki hub using https://technyon.github.io/nuki_hub/ once again, connected it to Wi-Fi and reconfigured everything. In the end: nothing changed. Issue still exists.

2. I looked into the factory reset option... there's a soft option of putting all settings back to default and a ful factory reset.
While I don't like both of them, the factory reset is no option. Too many things need to get reconfigured, the lock is just embedded in too much stuff (KeyPad 2.0, all permissions for all persons with all their smartphones and the app and widgets on them).

3. I prepared another ESP32, thinking maybe the Bluetooth chip on the original ESP32 had some issues. Flashed it with Nuki hub, brought into Wi-Fi, configured everything (same whole process as in step 1) and was about to pair the smart lock with that new Nuki hub. When right in front of the smart lock for pushing the button for 5 seconds to initiate the pair process, I thought: well... how about removing the battery of the smart lock for a second time, this time maybe a bit longer?

Result:

- After roughly one minute I put the battery into the smart lock again.
- It calibrated itself (important note: which it did NOT when pulling the battery for the first/last time, which obviously was not long enough).
- I tried to control it using Nuki hub (still on the original ESP32): et voila - it works. 🎉 

Looks like the smart lock had some issues?

Interestingly there were no clues about that in the Nuki app AND everything else (except Nuki hub) was working, meaning
- local control
- communication with the KeyPad 2.0
- control with all smartphones

**I will monitor this for a while (including letting `rstbcn` set to `60`), at least a day or two before I hopefully can and will close this issue.**

--------

**Optimization section:**

I still don't know why this happened. Which next to spending few hours on this makes me feel quite uncomfortable.
And I want to be better prepared for the next "hiccup" of the smart lock or similar issues where detailled logging is useful. Therefore
1. in regards to...
> MQTT logging can help as long as the problem is on bluetooth side and the ESP stays on Wifi.

...please tell me: what exactly do I need to do to after ticking `Enable MQTT logging`? I'm still a newbie to MQTT. Will there be a specific topic with some magic debug data? Will it contain sensitive information or can it be provided here?

2. maybe you have other ideas on how to improve reliability or problem detection?

E. g., as mentioned: Wi-Fi was working, MQTT was working, the ESP32 was working, everything was *looking* fine from MQTT and HA side - except no status updates / control commands were sent from/to the lock.
--> Any idea how to detect this scenario on the Nuki hub side?

https://www.youtube.com/watch?v=Szl3yYJ7hl0
Replace "developers" with "reliability", that's my ultimate goal of contributing to this project here ;-)

technyon, 2023-02-15:
There's no magic about the MQTT logs. It duplicates (almost) all messages that are written to the serial line to an MQTT topic (maintenance/log). To get a log you need to record the history of that node, since it only holds the last message. This of course only works as long as the ESP is connected to Wifi, but it's usually a lot easier than getting logs via serial port.

The Nuki devices do have flukes. My opener didn't open the door when pressing the button. I first thought it was a hardware defect since I didn't change anything. Factory reset solved it ... how that can happen no idea. Running the (android) smartphone widget (not app) randomly opens the door from time to time ...

bcutter, 2023-02-15:
> To get a log you need to record the history of that node

...so in HA switching to the MQTT integration and listening for the `nuki_door_A/maintenance/log` topic would be sufficient, right?

Any words on

> E. g., as mentioned: Wi-Fi was working, MQTT was working, the ESP32 was working, everything was looking fine from MQTT and HA side - except no status updates / control commands were sent from/to the lock.
--> Any idea how to detect this scenario on the Nuki hub side?

?

technyon, 2023-02-16:
If the Wifi was still working, but the lock doesn't, it sounds like the bluetooth stack died as already mentioned. Enable "Restart if bluetooth beacons not received", I think 60 seconds is a reasonable value.

Apart from that it's really hard to tell what's going on. If that doesn't help I can give a debug version which logs a ton of bluetooth debug data - then I'll have to ask Jeroen who maintains the NUKI ble library to look through it. It's just hex numbers not sure how he does it :D.

bcutter, 2023-02-16:
Alright. Nuki hub still working with the smart lock after ~ 24 hours. I'll close this issue for now - better re-open stuff in case it happens again.

For now I have and keep enabled these helpful settings:
- `Restart on disconnect`: true
- `Number of retries if command failed`: 5
- `Restart if bluetooth beacons not received (seconds; -1 to disable)`: 60

closed 2023-02-16 by bcutter