technyon/nuki_hub #128: Last Lock Action and Last Lock Action Trigger

magior, 2023-02-22:
When querying for keyturner states the lock return also: 
Last Lock Action
Last Lock Action Trigger

I was wondering if it is possible to publish these data via MQTT. It is the only way I've found to intercept a Lock and go action triggered by the button and use it on an automation.

technyon, 2023-02-22:
Hi,

The trigger is already published, the last lock action can certainly be added.

technyon, 2023-02-24:
Hi,

What do you need this for?

Some  thoughts about this: Yes, there's a flag for last lock action trigger, but it's always the same as trigger as far as I could see, so there's no point publishing it. The last lock action can be deduced from the lock state. If the state is locked, you know the last action was lock, and so on.

For the opener this I'd says this is broken: activateRTO does show up properly as last lock action, but sending deactivateRTO returns a last lock action of 0x00, which is an undefined value.

magior, 2023-02-25:
Thank you for your answer. I switched this week trying to use this very interesting project; before I used Nubli, I'm trying to reproduce all functions I have implemented with it.

Last lock action, in my experience and according to documentation, could also be LOCK AND GO, and I cannot deduct the status from the actual status. So, as an example, if Lock action is locked, last lock action is "Lock and go" and Last lock action trigger is "Button" I can trigger some automation that I don't want to be triggered when the lock status is triggered only by one button press, manually o automatically by nuki.
I hope this example could help you understand my use case. If I'm leaving, I use "Lock and Go" to trigger that, but if I want to close without leaving  I press the button, or I lock manually.

I noticed, after your message, that trigger and last action trigger seems always to be the same, but they shouldn't. Maybe a bug in Nuki firmware?

Marco

technyon, 2023-02-25:
I've added lastLockAction already for the lock, it'll be in the next release. How are trigger and lastLockActionTrigger supposed to behave?

magior, 2023-02-25:
Thank you for adding it.

Your messages have raised some doubts for me. The two values are always the same.

From the API documentation, "trigger" indicates what caused the state change, Last Lock Action Trigger is related to Last Lock Action. Maybe it is correct that they are always the same.
We could ask to the nuki developer forum, if needed.

technyon, 2023-03-08:
Hi, does the last lock action and the regular trigger work for you. If so I'd close the issue.

technyon, 2023-03-18:
Closing due to inactivity.

closed 2023-03-18 by technyon