technyon/nuki_hub #129: Lock state undefined

thor9898, 2023-02-27:
The hub pairs with the lock pretty quick but it keeps saying lock state: undefined

 [image]

Its been like that for about 3-4 days

My config:
NUKI Hub version: 8.11
run: true
deviceId: 1010125340
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
nwhw: 1
nwhwdt: 26
rssipb: -1
hostname: nukihub
nettmout: 5
restdisc: true
resttmr: -1
rstbcn: 10
lockStInterval: 10
configInterval: 10
batInterval: 30
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: -1
hasmac: true
macb0: -97
macb1: 53
macb2: 10
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock PIN set: No
Network device: Built-in Wifi
Uptime: 8 minutes
Heap: 65028
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

thor9898, 2023-02-27:
Is there anything i should be aware if i register it as app with out having the bridge? Its the only way i was able to short this problem out.

technyon, 2023-02-27:
This has nothing to do with the register as app setting. If the lock state is "undefined", most likely the ESP couldn't query the state from the lock because bluetooth communication failed. Why it didn't work is hard to say, thing you can try:

- Make sure the ESP has good reception, place it close to the lock if possible.
- I can see your lock PIN isn't configured. Although this isn't necessary to query the lock state, I'd recommend to configure it
- Unpair and re-pair the lock
- Factory reset the ESP32 and try again
- Try a different ESP32 module

thor9898, 2023-02-27:
i have it like around a meter from the lock and with no walls in between, so reception shouldn't be a problem.
i couldn't see any info about lock pin in the documentation, what's it for? how does it work?
Either way, unless you recommend me to change it I will keep it registered as app as it seems to be working fine.

technyon, 2023-02-27:
It's the PIN you set in the smartphone app. Certain functionality is only available when the correct PIN is provided to the lock, for example changing the configuration or adding / removing keypad codes.

So this means it's working now, or do you still have an undefined lock state?

thor9898, 2023-02-27:
by the smartphone app you mean the nuki official app or the web interface for the nuki hub?
now its working well, but only because I registered as app, I unliked the lock and tried to link it again and it wasn't pairing until I set the hub to register as app.

technyon, 2023-02-27:
The official NUKI App. It asks you to setup a PIN after pairing. This PIN can be configured in NUKI Hub. Setting it in NUKI Hub only provides it to the firmware, it doesn't configure it on the lock.

thor9898, 2023-02-27:
i didn't set up the pin in the nuki app, so it shouldnt be a problem

technyon, 2023-02-27:
ok, if all works I'd the issue

closed 2023-02-27 by thor9898