technyon/nuki_hub #130: Olimex POE board on crash loop

ruimarinho, 2023-02-28:
Hi,

I'm testing out the new 8.11 firmware with support for the Olimex hardware and after initially setting up the device via WiFi, I've selected the Olimex hardware option in the UI.

Even though the device has acquired an IP via PoE at least once, it has started a crash loop. This is what I was able to fetch under `/info` before entering the loop:

```
NUKI Hub version: 8.11
run: true
deviceId: 123456789
mqttbroker: 
mqttport: 1883
mqttuser: 
mqttpass: 
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
nwhw: 4
nwhwdt: 26
rssipb: -1
hostname: nukihub
nettmout: -1
restdisc: false
resttmr: -1
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: 98
macb1: 54
macb2: -75
MQTT connected: No
Lock firmware version: 
Lock hardware version: 
Lock paired: No
Lock PIN set: -
Network device: Olimex LAN8720
Uptime: 2 minutes
Heap: 94148
Restart reason FW: UnknownNoRestartRegistered
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
```

Eventually, after 15 minutes, the crash loop stopped automatically. I have no idea why/how.

Today, in the morning, I checked `/info` again and to my surprise the uptime was 48 minutes with the following crash reason:

```
Restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
```

Apart from the network hardware change, the nukihub has not been configured at all (no MQTT, no pairing, etc), so something is already crashing the device. Let me know how may I help debugging these issues.

ruimarinho, 2023-02-28:
Just checked again:

```
Uptime: 36 minutes
Heap: 57292
Restart reason FW: UnknownNoRestartRegistered
Restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
```

technyon, 2023-02-28:
Hi,

I've connect my Olimex board, it's up for 31 minutes now, I'll let it run a few more hours. If you have lock connected, maybe disable it so the ESP doesn't try to pair constantly.

Apart from that maybe check your setup. Is the board ok, does it get enough power? I have no POE setup, so I'm powering via USB, can't really verify if that makes difference but I doubt it.

technyon, 2023-03-01:
I'm at over 1200 minutes uptime now. Were you able to check your setup?

ruimarinho, 2023-03-01:
Just checked and still at 37 minutes uptime, so it's still crashing (I haven't changed anything so I guess it's expected).

1. Can I pair multiple nukihubs to a single Nuki Lock? I'm reluctant in removing my known-good ESP32 while testing this.2
2. If the answer is no, do I need to manually unpair the old nukihub via the UI or can I simply press and hold the Nukilock button?

technyon, 2023-03-01:
You can if you check "Register as app" before pairing. If you don't check it, the lock will automatically kick out your old ESP when you pair the other. It is possible though that one of the devices misses a lock state update, but for testing this should be fine.

ruimarinho, 2023-03-01:
Can I have two apps registered (iOS + Olimex)?

technyon, 2023-03-01:
Yes no problem. It's just a flag and it's purpose is to make sure there are never two bridges connected at the same time. You can get around that with NUKI Hub.

technyon, 2023-04-04:
@ruimarinho Do you still have the issue?

ruimarinho, 2023-05-21:
I have reflashed with 8.23 and now it seems to be working (but with new issues).

closed 2023-05-21 by ruimarinho