technyon/nuki_hub #134: Keypad 2.0 not pairing

Dielee, 2023-03-04:
First, thanks for this awesome work!
It works very nice, thanks a lot!

But, as in 8.14 Keypad 2.0 is now supported, I can not pair it...
Pressing the < button on the remote, and starting the pair process, seems not to work.
Log says `Nuki opener start pairing`, but paring isn't working.
I have also tried to insert the keypad code in "credentials", without success.

Do you have any advice ?

technyon, 2023-03-04:
The keypad isn't paired directly to the ESP. You connect the keypad using the smartphone app to the lock or the opener. When connected, you can enable keypad support in the web interface, and a new tree "keypad" in mqtt appears.

Dielee, 2023-03-04:
Ah okay! This works. But why is the log steady saying `Nuki opener start pairing` ?

technyon, 2023-03-04:
You have to pair the opener ... press the button on the opener for ~5 seconds to enable pairing mode on the opener.

Dielee, 2023-03-04:
You mean the < button ?

technyon, 2023-03-04:
What devices are you using? A keypad and an opener, or a lock?

Dielee, 2023-03-04:
A 2.0 keypad and an 3.0 Lock

technyon, 2023-03-04:
The keypad is not an opener. This is an opener:

https://nuki.io/en/opener/

The message about pairing is there because the ESP tries to connect to an opener, which you have to explicitly enable, making me think you want to use the keypad with an opener.

Go to the web interface and disable "NUKI Opener enabled" to make the message go away.

Dielee, 2023-03-04:
Ahhhh, sorry. Opener and keypad confused me. Sorry for this!!!

Thanks a lot!

closed 2023-03-04 by Dielee