technyon/nuki_hub #135: What's the real difference between "Bridge" and "App" registration?

binderth, 2023-03-04:
What's the benefit of "Bridge" vs "App" or vice versa? I think recommended is `Bridge` mode, but I'm not sure, what the pros and cons are for those?

From what I can see, the Nuki_Hub can't act as a Nuki Bridge for the purpose of the App getting SmartLock information also, if not connected to the local WiFi? correct? Meaning, using the Nuki App only works, if the phone is within bluetooth reach of the SmartLock.

technyon, 2023-03-04:
If you look at the API documentation, when pairing a device you can specify what kind of device it is:

0 … App, 1 … Bridge, 2 … Fob, 3 … Keypad

If you check register as app, NUKI Hub pairs as App, otherwise as Bridge. The only difference is that when a bridge is already registered, and you pair a new device that's also a bridge (or claims so as NUKI Hub does), the old bridge is removed. This is to prevent two bridges being paired at the same time. 

I'm not exactly sure what your second question is about, but NUKI Hub doesn't provide cloud functionality. You need some kind of home automation system (Home Assistant, iobroker, Openhab, ...), which your smartphone would access to trigger actions via NUKI Hub. If you want remote access to your lock, you'd need to get remote access to your home automation system somehow, like using a VPN to connect to your home network.

binderth, 2023-03-04:
Thanks for the explanation. 

I have a Nuki Bridge (which somehow throws massive 500er http-errors lately), and those if paired with a Smartlock allow the Nuki App to connect via the bridge even if not in BT range:
[image]

That's a nice touch, but as you said, not necessarily needed. I'm using openhab, so I can easily integrate the MQTT-topics to use it remotely, if needed.

technyon, 2023-03-04:
Yes that's the idea. It's also not really recommended to run the bridge and nuki hub in parallel, it can cause one or the other to miss lock state updates ... unfortunately. So if you switch, it's best to not use the bridge.

binderth, 2023-03-05:
yes, I unplugged the Bridge. I'm not sure, if the Bridge has an hardware defect as Nuki thinks - or it's some other bug. But I had this year alone 430 successful door actions via the bridge - and over 730 unsuccessful ones, where the bridge would just throw an 500 error. (I log every action and automatically retry, if an error occurs, or nothing happens)

using nuki_hub for two days now - and it's fast, reliable and no unsuccessful try! Thanks for the good work - and lightning fast answers here! ❤

closed 2023-03-05 by binderth