technyon/nuki_hub #136: New entity "binary_sensor.name_door_sensor" (state unknown) in Home Assistant after updating to Nuki Hub 8.15

bcutter, 2023-03-04:
Updated from Nuki Hub 8.10 to 8.15 looking at the KeyPad 2.0 support coming with Nuki Hub 8.14.

**After updating, a new entity in HA showed up: `binary_sensor.name_door_sensor` with state `unknown`.**

As the smart lock has a KeyPad 2.0 connected but not a door sensor, I wonder where this is coming from and if it's a bug.

It has to do something with the KeyPad 2.0 on that smart lock (3.0 PRO), as the other one (3.0) does not have a KeyPad connected to it - and it also does not have the ghost door sensor entity.

[image]

Before I just disable that entity in HA, I wanted to report this issue.

technyon, 2023-03-08:
Will be fixed in the next release.

bcutter, 2023-03-11:
After updating to 8.17 I
- got new entity `binary_sensor.door_keypad_battery_low`
[image]

- still see the confusing (as there still is no Nuki door sensor) `binary_sensor.door_door_sensor` (with state `unknown`)
[image]

TBH I didn't know what to expect when having KeyPad 2.0 support, at least the battery level in percentage would have been nice.
Anyway, the faulty door sensor entity still exists. MQTT integration has been reloaded.

I just disabled it once again, no big thing for me.
So in case you can confirm...
1) yes, the battery low entity is the only one provided for the KeyPad (2.0) and
2) the door sensor entity won't go away/is supposed to exist even there's no sensor attached to the Nuki
...I'm totally fine to close this.

technyon, 2023-03-11:
Please delete the auto discovery topic for the door sensore manually with using an mqtt client like MQTT Explorer. They API doesn't provide a battery level for accessories like the keypad or the door sensor, only a critical state.

bcutter, 2023-03-11:
I have no idea where to find that one. Checked the whole tree, found nothing. Still not very familiar with MQTT by the way -.-

[image]

technyon, 2023-03-11:
Check under "homeassistant" which is located under the root node.

bcutter, 2023-03-11:
Also there I can't find anything with `auto`, `discovery` or `autodiscovery`. Even searching the whole MQTT instance returns no results.

technyon, 2023-03-11:
Everything under "homeassistant" is auto discovery. The door sensor is a binary_sensor.

bcutter, 2023-03-11:
Jap, found it meanwhile. So you want me to delete this node ("topic"), right?

[image]

technyon, 2023-03-11:
yes

bcutter, 2023-03-11:
Done. Entity disappeared immediately from HA entity registry.

closed 2023-03-11 by bcutter

Larx, 2023-03-30:
Is it possible that this change is the reason that I have lost my (existing!) door sensor in HA since a few days?

technyon, 2023-03-30:
If it's paired with the lock it should appear in HA. The discovery topic is only deleted if no door sensor is paired.

Larx, 2023-03-31:
Thanks for the quick reply. I just noticed that my door sensor is missing from HA since a few days (running 8.18, door sensor still available in Nuki app) and this looked like a possibly related issue. Updating Nuki Hub did not yet fix the issue.