technyon/nuki_hub #137: Does /lock/query/lockstate also update /lock/doorSensorState?

binderth, 2023-03-05:
normally I guess, the states should be up-to-date, but if there's no change for some time, I'd like to trigger my smarthome for queries on the lockState and the doorSensorState, just to be sure. I'm not good at programming, especially C++, so does sending "1" to /query/lockstate potentially update both the lockState *and* the doorSensorState?

...and I figured, if doing so, nuki_hub only sends out a new message, if there was a change to the previous state I guess? I change the messages for both topics and the query/lockstate didn't result in getting a new message sent out? I think, if you specifically requested an update to something, you should get a new message? even if it's retained?

technyon, 2023-03-05:
If you check the bluetooth API:

https://developer.nuki.io/page/nuki-smart-lock-api-2/2#heading--keyturner-states

Everything that's in "Keyturner States (0x000C)" is updated. That includes the door sensor.

binderth, 2023-03-05:
wow. that was fast! again!
What's your thought on sending query information regardless, if changed?

From what I see, /query/config works that way already? /query/lockstate and /query/battery don't?

technyon, 2023-03-05:
Yes, there are checks to prevent sending updates if there's no state change. That's only logical, if nothing changed, no need to update it.

What's your use case for this?

binderth, 2023-03-05:
Use case is simply to be perfectly sure, that I´m working with the latest states.
I know, the topics are retained, but that's somehow the issue here! ;) 

Let's say, there's no change for quite some time. Which could be fine, because nobody opened the door. or there's some miscommunication going on, which I didn't noticed.
So, if there's no change for a defined interval, I'd like to query the lockState and doorSensorState - and I'd like to see an answer to my active query in the requested topics. So I can update my smarthome with that information: everythings fine!
or: if there's no answer to my query I can send me some kind of alert, that there could be something wrong here.
This can be achieved via two ways: sending out messages to the topics (even if nothing changed) or filling the "completionStatus" (or some other topic like "queryStatus") with an success or error message.

Why do I ask this? I'm bit of a "burnt child", as my Nuki Bridge went unreliable - but the Bridge at least threw out http codes, so I could identify, if there was an error. And the Bridge also responded with the respective states.

technyon, 2023-03-05:
Returning a status sounds much cleaner than guessing the success implicitly by retransmitting the same value

technyon, 2023-03-05:
Thinking about it, if it's about door sensor and lock state, this is the commandResult. Config is seperate, but it doesn't change often and I think it's not as important as lock and door sensor state ... like if you have the wrong LED brightness level in MQTT it's not a security issue.

You can also set the number of retries when the command fails, so it's quite unlikely that this is not updated.

technyon, 2023-03-05:
Correction: commandResult is actually something else, but you can set the number of retries and the ESP will try several times to query the state.

binderth, 2023-03-06:
> You can also set the number of retries when the command fails, so it's quite unlikely that this is not updated.

Are the number of retries logged in /lock/log? Then it would certainly be a thought to retry like 5 times and parse the log for retries to get back to possible interferences.

> Returning a status sounds much cleaner than guessing the success implicitly by retransmitting the same value

That's an architectural decision! 😎
But from what I see /query/config and /query/battery fill the corresponding topics no matter what - even if the values didn't change. So if you're aiming on consistent architecture, I'll suggest going one path consequently... 😉

technyon, 2023-03-06:
Check release 8.16

closed 2023-03-08 by binderth