technyon/nuki_hub #139: MQTT

Asurinx, 2023-03-06:
Hello and sorry if this is not properly posted, im stucked on the MQTT step, how i can obtain that adress exactly? if if the IP assigned to my ESP 32 i have it

technyon, 2023-03-06:
Hi,

Do you have an an MQTT broker running? If so, you need to enter the IP address of that broker, it's not the same es the ESP address. The ESP is a client to the broker.

Asurinx, 2023-03-06:
Im just in the step of opening it in my computer via the adress assigned to the ESP32, where i see

MQTT and Network Configuration
NUKI Configuration
Credentials
Firmware update
WiFi

technyon, 2023-03-06:
Please tell me more about your setup. NUKI Hub only makes sense if you use some kind of home automation, like home assistant, iobroker, openhab ... do you have such a system?

Asurinx, 2023-03-06:
Was trying to use it as bridge for the Nuki Lock 3.0 , and be able to use it when im outside if needed, not possible via the app?

technyon, 2023-03-06:
On it's NUKI Hub can't do that. There's no cloud functionality like the bridge has, and it's not connected to the smartphone app.

You need to install a smarthome system on a server like this:

https://www.home-assistant.io/

Then connect NUKI Hub to the smarthome system.

Asurinx, 2023-03-06:
Oh ok sorry! im first time on this and didnt see in any part, just installed it last friday and i wish to have more control when im out of home and see this project :)

technyon, 2023-03-06:
Maybe get yourself a Raspberry Pi and try home assistant:

https://www.home-assistant.io/installation/raspberrypi

Asurinx, 2023-03-06:
I have bought the esp32, isnt this enought?

technyon, 2023-03-06:
No an ESP alone is not enough. It has to be connected to a some smarthome system, like home assistant.

technyon, 2023-03-06:
You can use:

- a Raspberry Pi (cheap option)
- any computer running linux or windows (expensive option; Intel NUCs are often preferred)
- anything that runs MacOS (the very expensive option)

Before you Raspberry, be aware that you have to configure those systems. Remote access doesn't come out-of-the box and you'll have to either configure a VPN or use some kind of cloud.

Asurinx, 2023-03-06:
Woah! i was trying to save the cost of the bridge but the raspberry price is highter than that... i think i got a fail on my planification haha

closed 2023-03-07 by technyon