technyon/nuki_hub #140: Help Wanted: How to get Bluetooth signal strength sensor working

thor9898, 2023-03-07:
I've had this working for over a month now, but the Bluetooth signal strength has always been unknown and I can't find how to get it in the readme.md
Same happen with the WiFi signal strength, although this is not as important as the other.

[image]

alexdelprete, 2023-03-07:
What version of HA and Nuki Hub?

Go in HA developer tools and post these two screenshots of the two sensors:

[image]

[image]

Also, check MQTT data in the broker, and post the contents of these two topics highlighted in yellow:

[image]

thor9898, 2023-03-07:
@alexdelprete Sorry i didnt provide the versions earlier, I have the latest release of HA and Nuki Hub.

[image]
[image]

I don't know how to get to that last pic you uploaded, I'm very bad at mqtt. I have mosquitto broker addon, so if you guide me ill upload it asap

I don't know if this would work:

[image]

[image]

alexdelprete, 2023-03-07:
> I don't know how to get to that last pic you uploaded, I'm very bad at mqtt. I have mosquitto broker addon, so if you guide me ill upload it asap

You need a tool like MQTT Explorer to navigate the MQTT broker. Install it and configure it.

Anyway, I already see some strange things: the nuki/lock/rssi and nuki/lock/wifiRssi topics have zero received msgs:

[image]
[image]

This means something not right at the broker level. Can you please show me how you configured the **MQTT NUKI Smartlock Path** on the Nuki Hub web UI?

[image]

alexdelprete, 2023-03-07:
Also, please show me the subtopic of the lock config:

[image]

capandcamera, 2023-03-07:
I resolved my bluetooth one by changing the following setting in the MQTT & Network section

```RSSI Publish interval (seconds; -1 to disable)

My wifi still doesn't seem to publish an RSSI though but hopefully this helps you fix your bluetooth one atleast.

thor9898, 2023-03-07:
@alexdelprete 
Subscribed topics:
nuki/battery/critical
10 most recently received message(s)
Received 10:42:02
QoS: 0
Payload: 0
Received 10:43:39
QoS: 0
Payload: 0
Received 10:45:40
QoS: 0
Payload: 0
Received 10:47:43
QoS: 0
Payload: 0
Received 10:50:19
QoS: 0
Payload: 0
Received 10:51:42
QoS: 0
Payload: 0
Received 10:53:52
QoS: 0
Payload: 0
Received 10:55:44
QoS: 0
Payload: 0
Received 10:56:06
QoS: 0
Payload: 0
Received 16:37:19
QoS: 0
Payload: 0

thor9898, 2023-03-07:
@capandcamera Oh wow, that made it!! Even wifi is showing no!!!

[image]

closed 2023-03-07 by thor9898

alexdelprete, 2023-03-07:
> @capandcamera Oh wow, that made it!! Even wifi is showing no!!!
> 
> [image]
> 
> Thank you very much!!

You had it disabled (-1)?

thor9898, 2023-03-07:
@alexdelprete yes, as I said I couldn't find info about how to enable it in the info.
Sorry if I took much of your time.

alexdelprete, 2023-03-07:
> yes, as I said I couldn't find info about how to enable it in the info.
> Sorry if I took much of your time.

Oh no, I'm sorry because I didn't tell you about that basic setting first, I assumed you had already tried basic things. Bad assumption, sorry. :)

thor9898, 2023-03-07:
@alexdelprete Yes it's updating. Now everything works fine right now :)
Thanks for your willingness to help me, I appreciate it, my Mediterranean brother :)

alexdelprete, 2023-03-07:
"Puerta de casa..." leads me to think you're from Spain. :)

I'll reveal you one secret: I was born in Madrid. My father worked there for some years when I was a little kid. But I'm 100% italian (from Napoli). :)

capandcamera, 2023-03-08:
> > My wifi still doesn't seem to publish
> 
> did you check data in the broker? if it is updating...

I can't see it at all in MQTT Explorer.
RSSi now appears under: nuki/lock/rssi
but I can't find wifiRssi anywhere (checked /lock and /maintenance

/maintenance only has 'uptime' and 'reset'

alexdelprete, 2023-03-08:
Are you sure you're looking at the right nuki_hub topic in the broker? Did you change it in the options maybe? The default root topic is nuki_hub if I recall correctly, but it's customizable.

capandcamera, 2023-03-08:
I cleared down the topic and restarted nuki_hub - The topic has been set to 'nuki' in config.
I just enabled MQTT logging to see if it was that causing some options to not appear, i can see log has appeared in the right topic but I do not get as much information as you have in your maintenance topic.
[image]

my uptime in the maintenance topic also stays at 0 regardless of uptime - Is there a setting that i need to configure to allow maintenance reporting?

alexdelprete, 2023-03-08:
You're using latest version, 8.16? This is really strange...here's a bit of my config in MQTT section:

[image]

capandcamera, 2023-03-08:
Settings are very similar, other than the network hardware section which i've left as was.
[image]

alexdelprete, 2023-03-08:
You have the W5500? Maybe that's the difference...but let's wait for @technyon to step in.

capandcamera, 2023-03-08:
Weirdly I don't, 
I'm using the Atom Lite (Non POE version) via Wi-Fi.

mundschenk-at, 2023-03-08:
If `uptime` stays at `0`, you very likely have a reboot loop.

capandcamera, 2023-03-08:
Thanks @mundschenk-at you were correct!  I've noticed 'Restart timer (minutes; -1 to disable)' was enabled to 1 minute. 
I'm not sure when this setting was changed but it then caused it to reboot before it managed to get a WifiRssi.

The topic is now showing and uptime is increasing!

alexdelprete, 2023-03-08:
You said settings were similar to mine. :)

Fix the network hw setting. I can't understand how it is connecting, unless it fallsback automatically to wifi. Anyway you should set it at Wifi only.

capandcamera, 2023-03-08:
> You said settings were similar to mine. :)
> 
> Fix the network hw setting. I can't understand how it is connecting, unless it fallsback automatically to wifi. Anyway you should set it at Wifi only.

I thought they were, i missed the darn - on that option. 

I'll correct the networking options now though!

mundschenk-at, 2023-03-08:
> Fix the network hw setting. I can't understand how it is connecting, unless it fallsback automatically to wifi. Anyway you should set it at Wifi only.

I am pretty sure it does exactly that: It probes the given GPIO and falls back to WiFi if it doesn't detect the ethernet hardware.

alexdelprete, 2023-03-08:
Probably, but since he has a wifi M5 Atom Lite...that setting is wrongly configured.

technyon, 2023-03-08:
This option actually is the old method of detecting a W5500. If you connect ground to GPIO 26, it switches to W5500, otherwise it uses Wifi. And yes there's fallback Wifi if no W5500 is present. I'm actually thinking about removing the detect via GPIO logic, since there's also LAN8720 now and it doesn't really make sense anymore.

thor9898, 2023-03-10:
I know this might not be the place to ask it but as it's a probably easy question for you guys I think it's easier to ask it in this already open issue than opening a new one only for this silly question.
In the system information where is the contact sensor status?

technyon, 2023-03-10:
@thor9898 Please update to 8.17

thor9898, 2023-03-10:
@technyon done already, should I be able to find it now?

technyon, 2023-03-10:
I think I misread your question. 8.18 fixes a bug about home assistant auto discovery and the door sensor. Currently the door sensor doesn't show up in the system information page.