technyon/nuki_hub #142: 8.16 update

gatd, 2023-03-07:
8.16 ota update fails to update stays on 8.15

technyon, 2023-03-07:
Are you sure you have the right binary? I just did an upgrade from 8.15 to 8.16 and it says 8.16 now.

gatd, 2023-03-08:
yes its the right binary I have re downloaded and tried several times.

technyon, 2023-03-08:
It seems I forgot to update the webflash binary, I've updated it to 8.16.

So can you pair the lock?

gatd, 2023-03-08:
I spent all day trying everything in the end  I used flash download tools earsed the flashed with that and now showing 8.16 !

technyon, 2023-03-08:
It's very strange, for some people it just works, other struggle with OTA. I have several ESPs, and it's almost always first try that it works.

magior, 2023-03-09:
I'm experiencing the same issue upgrading from 8.13. 
After OTA upgrade it stays on 8.13.

technyon, 2023-03-09:
There are some fixes for OTA in 8.15. You can update via serial to 8.16, just don't erase before flashing, and your settings should be kept.

gatd, 2023-03-09:
Ok I’m getting door lock sensor unavailable in home assistant with 8.16

technyon, 2023-03-09:
Please post the system information page (click on the firmware version number)

gatd, 2023-03-09:
NUKI Hub version: 8.16
run: true
deviceId: 2889820003
mqttbroker: 192.168.0.11
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwhwdt: 26
rssipb: 360
hostname: nukihub
nettmout: -1
restdisc: false
resttmr: -1
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: -1
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 2.14.5
Lock hardware version: 12.2
Lock paired: Yes
Lock PIN set: Yes
Network device: Built-in Wifi
Uptime: 134 minutes
Heap: 91228
Stack watermarks: nw: 6104, nuki: 640, pd: 244
Restart reason FW: UnknownNoRestartRegistered
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.

alexdelprete, 2023-03-09:
> Ok I’m getting door lock sensor unavailable in home assistant with 8.16

Door sensor works perfectly with 8.16 in HA:

[image]

Check the mqtt broker data (with MQTT Explorer) to see what value is in there for the door sensor:

[image]

technyon, 2023-03-10:
I figured it out: There were two variables swapped. Whenever you had a keypad, it would publish auto discovery for the door sensor and vice versa. Fixed in release 8.17.

gatd, 2023-03-10:
installed 8.17 started working again
thanks

closed 2023-03-10 by technyon

alexdelprete, 2023-03-10:
> I figured it out: There were two variables swapped. Whenever you had a keypad, it would publish auto discovery for the door sensor and vice versa. Fixed in release 8.17.

I have both keypad and door sensor, so I was seeing the states reversed? :)

gatd, 2024-11-12:
hi man
long time user of you Hub sw
I just uninstalled the latest nukihub its v9.0.1

But i have a problem with Homeassistant not seeing everything shows unavailable and greyed out no sensors
mqqt in HA shows nukihub messages being received

any help please?

thanks
kind regards

> On 10 Mar 2023, at 18:Hi man
usb

gatd, 2024-11-12:
here are my settings

System Information
------------ NUKI HUB ------------
Version: 9.01
Build: 11345224811.2.1
Build type: Release
Build date: 2024-10-15
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 3
Config version: 901
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 66388
Total internal heap: 260664
PSRAM Available: No
Network task stack high watermark: 8996
Nuki task stack high watermark: 5128

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.01
Allow update from MQTT: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.0.48
SSID: SBS
BSSID of AP: 40:0D:10:42:04:49
ESP32 MAC address: C0:49:EF:F9:9E:18

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
Fallback to Wi-Fi / Wi-Fi config portal disabled: No
Connect to AP with the best signal enabled: Yes
RSSI Publish interval (s): 60
Restart ESP32 on network disconnect enabled: Yes
Reconnect network on MQTT connection failure enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.0.11
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT lock base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons recieved: -1

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: Yes
Publish Nuki device config: No
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: Yes
Keypad query interval (s): 1800
Enable timecontrol control: Yes
Publish timecontrol topic per entry: Yes
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.0.48

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 1247116004
Nuki device ID: ***
Firmware version: 
Hardware version: 
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: App

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

technyon, 2024-11-14:
@gatd Please use version 9.0. 9.1 was pulled due to some issues. We're on track for a 9.2 release though.