technyon/nuki_hub #144: Support for the LilyGO-T-ETH-POE

OmegaModr, 2023-03-12:
Hey,

loving your project, working great so far with 2 of my locks. I've setup two esp32 "mini" style modules, but sadly the location close to the locks (to get the best BT signal) is in a poor wifi coverage area. 

As I do have a rj45 jack close by and a poe switch would it be possible to support the https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE board?

No idea how close to the setup of the Olimex ESP32-POE it is, but hoping it isn't to hard to support it.

technyon, 2023-03-15:
You can try the attached binary. This is untested, since I don't have the device. I took their parameter set and implemented a new device type for this. You have to go to network settings and select the lilygo device first.

[nuki_hub-lilygo-1.zip](https://github.com/technyon/nuki_hub/files/10982980/nuki_hub-lilygo-1.zip)

OmegaModr, 2023-03-15:
Didn't expect you to be that fast :), the boards are still in the post I'll give them a go as soon as they arrive.

Kudos for supporting this.

OmegaModr, 2023-03-24:
Hi, can confirm that the firmware is working, mind including it in your regular builds?

I'll leave it running if it encounters any issues

technyon, 2023-03-24:
Check Release 8.19.

closed 2023-03-24 by technyon

Diabl0570, 2023-10-26:
For me (Lilygo POE v1.1) even if I select Network hardware: LilyGO T-ETH-POE in the info page i see that its stilll using WIFI: Network device: Built-in Wifi. I tried the latest built and the build linked in your comment.

technyon, 2023-10-27:
That should do it. If LAN doesn't work, it falls back to Wifi. Can you try some other device on the LAN cable and see if it works? Also please post the content of the system information page (click the firmware version).

Diabl0570, 2023-10-28:
1) The LilyGO gets powered by POE
2) The LAN cable does work for my laptop
3 My system info is:

```
NUKI Hub version: 8.27
run: true
deviceId: 603285012
deviceIdOp: 603285012
mqttbroker: 
mqttport: 1883
mqttuser: 
mqttpass: 
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 7
rssipb: 60
hostname: nukihubpoe
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 
regAsApp: false
nrRetry: 
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: -65
macb1: -72
macb2: 36
MQTT connected: No
Lock firmware version: 
Lock hardware version: 
Lock paired: No
Lock PIN set: -
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 3 minutes
Heap: 89260
Stack watermarks: nw: 6184, nuki: 2296, pd: 240
Restart reason FW: NetworkDeviceCriticalFailure
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

technyon, 2023-10-28:
It's strange it should just work. Can you check if it works if you power via USB? I didn't have POE equipment at the time for testing.

Diabl0570, 2023-10-28:
Connecting it to USB first doesn't change anything. On the POE switch I can see the power led lighting up, but the LINK led never lights up, so there I can verify that the LAN connection never is initiated.

technyon, 2023-10-28:
I can't really tell what's going wrong. From the code side, I'm using the driver for the LAN interface and configuring it accordingly to how it's wired. If that doesn't work, I'd need to have the actual module to debug it.

Can you rule out broken hardware, for example by flashing some other firmware that uses the LilyGO LAN?

Diabl0570, 2023-10-30:
I can't seem to flash a new firmware, so maybe it is a hardware failure. Maybe I damaged the Lilygo while desoldering the TTL flasher. I ordered a new TTL flasher to make sure that isn't the broken component, I also ordered a Olimex ESP32-PoE-ISO to try that one. 

For now I will assume the Lilygo is damaged, if it turns out otherwise I will update here. 

Anyway thank you very much for your work done on this awesome project!

technyon, 2023-10-31:
Olimex is very solid hardware. Expensive but well built.