technyon/nuki_hub #146: request function: Restart Button or/and Pairing button

DaKoarli, 2023-03-18:
Hello,

i am searching for a button to make an manual restart of the esp. i have mounted it in a box which is not easy to reach. It would be great if there is a Restart button on the configuration interface.

Also for pairing it would be easier to have an restart or pairing button without to hardreset the esp.

If there is already a function like that then i am sorry i haven't found yet.

Nice regards and thank you for your great work

Dominik

(Sorry for my bad english)

technyon, 2023-03-18:
In mqtt please check under "maintenance/reset". If you set it to 1, the ESP will reboot.

If you unpair a device (which is possible via the web interface), pairing mode will automatically start again.

DaKoarli, 2023-03-18:
thank you for your fast response.

the last few times when i did a unpair it only would pair again if i was reseting the esp, but maybe it was a problem on my side thank you for the information.

closed 2023-03-18 by DaKoarli