technyon/nuki_hub #147: Nuki lock state undefined after a couple of days

magior, 2023-03-18:
I've version 8.18,  after 3/4 days running without porblems the Nuki lock seems to stop working.
On the web page, I found the lock state as "undefined". A soft reboot doesn't solve the problem, and accessing boot log I find these:

[07:30:54]MQTT connected
[07:30:59]Nuki start pairing
[07:30:59]Nuki paired
[07:31:09]Querying lock state: Querying lock battery state: failed
[07:31:14]Reading config. Result: failed
[07:31:19]Reading advanced config. Result: failed

Unplugging power solves the problem.
Is there something I can do for better debugging? Is there a way to make a hard reset if the nukihub isn't plugged to a PC?

technyon, 2023-03-18:
Not sure what happened, but I think your lock lost the pairing to the ESP. Unpairing and re-pairing should help.

technyon, 2023-03-25:
@magior Any update on this? Did re-pairing help?

magior, 2023-03-25:
Re-paired as a bridge (It was paired as an app before) and has been working in the last five days.

Still on 8.18, and now I'll try to upgrade to 8.19.

magior, 2023-03-25:
Re-paired as a bridge (It was paired as an app before) and has been working in the last five days.
Just now, after my first release of this message, I had the same issue and the state is "undefined" from the web interface. 
A power adapter powers the ESP32 and I cannot read logs. Is there a way to read them without a USB cable attached?
If I reboot the ESP32 it starts working again.

Still on 8.18, and now I'll try to upgrade to 8.19.

technyon, 2023-03-25:
You can set the "maintenance/reset" MQTT node to 1 to reboot the ESP. There's actually an issue in the bluetooth stack which causes it to fail, unfortunately I have no control over this since it's outside of my code. There's a work around, you can enable "Restart if bluetooth beacons not received". It's safe to set it to a low value like 60 seconds, since the ESP reboots within seconds.

If that doesn't help, there are a few more workarounds you can try:

- Monitor the lock state. Whenever it shows undefined, automatically write "1" to "maintenance/reset" to trigger a reboot.
- Enable "Restart timer" and set it so something like one or two days. This will reboot the ESP in regular intervals. Not the best solution if the problem occurs randomly

magior, 2023-03-25:
Thank you, I'll try the reboot via MQTT if it's different from what happens with "Restart if bluetooth beacons not received". I've already configured it, and it doesn't solve. 
From my tests, only a cold reboot (unplugging and replugging from power) solves it.

magior, 2023-03-29:
I tried rebooting via MQTT. After reboot, the state shows as "undefined". 

On logs I have:

MQTT connected
Nuki start pairing
Nuki paired
Querying lock state: Querying lock battery state: failed
Reading config. Result: failed
Reading advanced config. Result: failed

So, nothing changed. My last reboot (unplugging power) was 4 days ago.

technyon, 2023-03-29:
Please

- Update to firmware 8.20
- Post the content of the system information page

Things you can try:

- Factory reset the lock
- Erase the ESP, then flash and re-pair

technyon, 2023-04-04:
@magior Did you try any of the suggestions?

magior, 2023-04-05:
Yes, I tried and waiting to see if the problem occurs again.
Unfortunately, tomorrow we will have a small programmed blackout and the ESP32 will restart (and I have to continue waiting).

I have also added a second device (paired as an app) because I suspect it could be a hardware issue.

Here the system information page:

NUKI Hub version: 8.20
run: true
deviceId: 733291356
mqttbroker: 172.18.1.122
mqttport: 1883
mqttuser: 
mqttpass: 
mqttlog: true
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 2
nwhwdt: 26
rssipb: 60
hostname: nukihub2
nettmout: -1
restdisc: false
resttmr: -1
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
gpiolck: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.4.10
Lock hardware version: 4.10
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: Yes
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 1 minutes
Heap: 89328
Stack watermarks: nw: 6104, nuki: 664, pd: 276
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_BROWNOUT: Brownout reset (software or hardware)

MrHollowPS, 2023-04-07:
I have the same issue unfortunately. I am on 8.20 i have not repaired yet.

@technyon what do you suggest i do first?

Dielee, 2023-07-30:
Same here, automatic reboot don't work. Mqtt reboot don't work, same as automatic reboot. Only a cold reboot by unplug the esp works for me. Is there anything we can do ?

technyon, 2023-07-31:
It's hard for me to say what going on there, it would need detailed debugging of the bluetooth communication. Enabling the restart if no beacons received options received fixed all problems when we were testing, but maybe this is another issue. I wish I could give better advice. Maybe check the hardware, the power supply, try a different ESP.

Dielee, 2023-07-31:
Okay. Sure I will create some logs. Just send me the firmware, I will flash it as soon as I can.

technyon, 2023-08-01:
Here you go. There will be a lot of debug output. You can trigger to query the lock state by setting the node "lock/query/lockstate" to 1, try to capture what happens after you do that.

[nuki_hub-8.25-dbg.zip](https://github.com/technyon/nuki_hub/files/12233392/nuki_hub-8.25-dbg.zip)

iranl, 2024-05-17:
Closing as this issue has been stale for more than half a year and the issues have been reported with an at this time 15 versions out of date Nuki Hub. Please reopen if this issue still occurs on 8.34 and post the contents of the info page as suggested by @technyon

closed 2024-05-17 by iranl