technyon/nuki_hub #148: Is it possible to switch a PIN of ESP32 low/high for showing Lock status?

BennoB666, 2023-03-18:
Is it possible in future release to have a PIN which state the status of the NUKI Lock?

technyon, 2023-03-18:
Well, there are more than two states, but it could be mapped like the binaryState for HA. What's your use case?

BennoB666, 2023-03-18:
Thx for fast answer. I have a second Door for my garage where is installed an electric door opener. I look for a solution to lock the garage door when i lock the nuki. I asked by nuki if it possible to lock two doors with one Keypad/app/... but that is not possible. So when it ist possible to trigger a pin of ESP i can break the wire to the electric door opener and nobody can access. In best case i can input a delay to trigger the pin or can change the pin via MQTT so node red can trigger the pin to the nuki state.
Also with this pin you can show the status of the door with a LED or simular.

technyon, 2023-03-18:
You use a NUKI lock for your door and a NUKI Opener for the garage?

BennoB666, 2023-03-22:
For me its enough to have such LED PIN. I can do this also with a second esp witch toggle the relay (i have to supply 12V to the opener for opening means 12V -> Open, 0V -> close) but the NUKI-HUB is 50 cm away from the Opener and for me it is easy to grab the PIN and toggle a transistor/relay. Maybe it is also good for opening/closing yard gate or Garage door. I can do this with Homeassistent when it possible to change PIN state via MQTT. I prefer NodeRed. Maybe you can implement to passthrough unused PINs of ESP to MQTT.
Thx a lot.

technyon, 2023-04-08:
Check release 8.22

victorigualada, 2023-04-14:
Hi,

First of all thanks for this amazing work. Was using HA addons to act as bridge but this solution is great. 

I'm trying something similar. Currently I have an [eWelink relay](https://es.aliexpress.com/item/4001351788406.html?spm=a2g0o.productlist.main.33.36e74fdewqPb1C&algo_pvid=f41511f2-fa60-4956-af9f-486d875b154d&algo_exp_id=f41511f2-fa60-4956-af9f-486d875b154d-16&pdp_npi=3%40dis%21EUR%216.19%214.95%21%21%21%21%21%40214528be16814674787987517d070d%2110000015774854539%21sea%21ES%210&curPageLogUid=Wds4pWoauHTv) to open my building's entrance (something similar to the Nuki Opener) behind the intercom. I can then from HomeAssistant trigger the relay and open the main entrance.

Would this be possible to accomplish with the current `nuki_hub` functionality? I'd love to remove that relay and use only the `nuki_hub` in the ESP32 for both things (Nuki Lock and intercom) for sake of space and energy costs. 

If it's not supported, is there a way it could be extended to enable one GPIO to be configurable in MQTT to trigger a high state with a particular topic? Maybe also with some kind of inching to hold the intercom opened for `x` seconds.

Thank you!

closed 2023-04-19 by technyon