technyon/nuki_hub #152: M5Stack Atom PoE connected via ethernet keeps connecting to WiFi

beloso, 2023-03-22:
I have a Atom PoE that is working brilliantly, however every so often to reverts to WiFi instead of maintaining the ethernet connection.

I would love to get more logs, but I don't know how xD

Is there a fix or workaround?

technyon, 2023-03-23:
I noticed the same problem every now and then, and I think I've found a bug in the connection logic. Please try this binary:

[nuki_hub-8.19-pre-2.zip](https://github.com/technyon/nuki_hub/files/11054969/nuki_hub-8.19-pre-2.zip)

beloso, 2023-03-23:
I've flashed it and it's using Ethernet for now. 
I'll monitor it over the next few days and report my findings.

technyon, 2023-03-24:
You can upgrade to 8.19, the fix is included

beloso, 2023-03-27:
In the Pre version and in the Final one the issue appears to be solved for me. I'm closing the issue.

Thanks for your hard work.

closed 2023-03-27 by beloso