technyon/nuki_hub #156: Configure MQTT presence/devices publish interval

danimaciasperea, 2023-03-26:
Add an option in MQTT and Network Configuration page to configure the publish interval of the MQTT presence/devices topic.

technyon, 2023-03-27:
What's your use case that you need to configure this?

bcutter, 2023-03-28:
Maybe it's updating too frequently (impact on battery life)?

danimaciasperea, 2023-03-28:
Yes exactly, it is being updated every 3 seconds. I don't know if it is a bug (I had problems with update frequency of other topics in a previous version). Battery should not be a problem, since this is independent from the look device, right?

bcutter, 2023-03-28:
> Battery should not be a problem, since this is independent from the look device, right?

My fault. Of course, it is done by the ESP, not the SL.

That's why I'm fine and don't care about the update interval.

Having a "only update once every XX seconds" option would still be nice to limit things, but not even in my personal top ten ;-)

technyon, 2023-03-28:
Correct, there's no impact on battery life, since the lock isn't involved in presence detection at all. The ESP just publishes the adresses of whatever beacons it receives. Three seconds isn't a bug, presence detection should be up-to-date. The only concern could be that it generates network traffic, but even that is very little.

iranl, 2024-02-29:
Should this be closed as not a bug / not being worked on / won't fix?

technyon, 2024-03-01:
Yes, forgot to close it.

closed 2024-03-01 by technyon