technyon/nuki_hub #157: Unable to add/update keypad code

kordys, 2023-03-26:
Hi, I am having a problem with adding/updating a keypad code using the MQTT. I followed the docu and ended up with this approach:
[image]

Can someone point me in the right direction, please?

Thanks a lot!

technyon, 2023-03-26:
Looks fine. You have to send the "action" last ... maybe don't choose a code like 666666. The lock has certain checks built-in, for example no code can start with "12", to prevent people from using 123456 as a code. For me, something like 334455 worked.

Also, did you pass the lock PIN to NUKI Hub?

kordys, 2023-03-27:
> Looks fine. You have to send the "action" last ... maybe don't choose a code like 666666. The lock has certain checks built-in, for example no code can start with "12", to prevent people from using 123456 as a code. For me, something like 334455 worked.
> 
> Also, did you pass the lock PIN to NUKI Hub?

Thanks for the answer. I did provide the PIN code to the NUKI Hub and the lock itself works as expected. Regarding the keypad codes - can you please elaborate on the statement "You have to send the "action" last...". I am providing the JSON as a whole and AFAIK the order doesn't matter. Or am I wrong? Thanks!

technyon, 2023-03-27:
I'm not really sure in which way you use JSON here, but the firmware doesn't accept JSON as input. Maybe this is specific to your smarthome solution, maybe elaborate a bit about your setup.

NUKI Hub subscribes to nodes in MQTT, the order you set those nodes is important: First set the parameters, then trigger the command.

kordys, 2023-03-27:
> I'm not really sure in which way you use JSON here, but the firmware doesn't accept JSON as input. Maybe this is specific to your smarthome solution, maybe elaborate a bit about your setup.
>
> NUKI [...]

Thanks for all the input. I finally managed to solve it. I didn't realize that those are separate topics and you need to publish each of them individually. Now it makes sense and everything works as expected :)

kordys, 2023-03-27:
Actually, I was able to add a keypad code, but I am unable to disable a keypad code :( Steps I performed:
1. topic **nuki/keypad/command/id** with <code id> payload
2. topic **nuki/keypad/command/enabled** with 0 payload
3. topic **nuki/keypad/command/name** with "<name>" payload
4. topic **nuki/keypad/command/code** with "<code>" payload
5. topic **nuki/keypad/command/action** with "update" payload

The keypad code stays enabled. Am I missing something, please?

Thanks.

technyon, 2023-03-27:
Please have a closer look at the readme:

`keypad/command/action: The action to execute. Possible values are add, delete and update`

You're updating the code, not deleting it. To delete a node, use the "delete" command. Also the only node you have to set is id, but this is not the path "code_id", but a plain number you read from the "id" node within the codes subtree.

technyon, 2023-03-27:
ok I see, so as far as I can see it should work then. Is it still not updating?

kordys, 2023-03-27:
> ok I see, so as far as I can see it should work then. Is it still not updating?

Finally, I managed to get it to work. My problem was that I encapsulated the values of the payload to quotes for string :) It works now as expected. Thanks for the help.

closed 2023-03-27 by kordys

kordys, 2023-03-28:
I am using Home Assistant, so right now I created a script to enable/disable keypad code that, based on parametrization, publishes all the topics with appropriate payload.

edobez, 2023-05-24:
@kordys, could you maybe share the script? I'm having troubles updating the the nodes.
I always get "error" in the response. I suspected it was due to a code name with a whitespace, but this wasn't the case.

edobez, 2023-05-24:
Here is what I'm using:
```
#!/usr/bin/env bash

name="User 1"
id="5"
code="376123"
enabled="1"

mosquitto_pub -t nuki/keypad/command/id -m $id
mosquitto_pub -t nuki/keypad/command/code -m $code
mosquitto_pub -t nuki/keypad/command/name -m "$name"
mosquitto_pub -t nuki/keypad/command/enabled -m $enabled
mosquitto_pub -t nuki/keypad/command/action -m update
```

technyon, 2023-05-25:
Are you sure you got the correct id? You read it from the "id" node, not the name of the parent node (e. g. not "code_5", but the content of "code_5/id"). I know it's a bit confusing, but hard to wrap into an mqtt tree otherwise.

edobez, 2023-05-25:
Yes, as you can see from the example, the `id` variable is only composed about the number, not the whole `code_10`.
I'm not sure if MQTT distinguishes between integers and strings, but I assumed that they are all strings, so the quotes won't matter.
In the example I just used the quotes around $name because it might contain whitespaces.

technyon, 2023-05-25:
What I'm saying is you can't use "10" if the parent node is "code_10". you have to look at the content of the "id" node that's under "code_10" and use that as id. It can be and is most likely different than 10.

technyon, 2023-05-27:
Hi, ok the id should be fine then, I just wanted to make sure.

"error" means for whatever reason the smartlock firmware didn't like some parameter that was sent to it. Can you get logs, mqtt logs should be fine in this case? Enable "Enable MQTT logging" on NUKI Hub side, on the broker side you need to get a history of what's published to the mqtt log topic.

edobez, 2023-06-05:
Hi, sorry for the delay. I think I'll be able to test it at the end of June, I don't have access to it now.
I'll post here again.