technyon/nuki_hub #158: Feature request: Create switch/button in HA for restarting NUKI Hub

bcutter, 2023-03-28:
Can we have an out of the box switch (or **button** meanwhile) entity for restarting the NUKI hub?

I had some situations where this was needed, physical access to the ESP was not possible and logging in to the web interface (open MQTT/network settings and click save to trigger restart) is only possible for me (due to the access protection) but still neither handy nor quickly.

Latest occurrence (which made me finally write this FR) by the way: SL died (Bluetooth died), pulled the battery. After that, NUKI hub did not reconnect automatically within few minutes - at least changing settings (LED off/on etc.) from HA had no effect on the SL. For that a quick restart sent from HA would just be perfect.

technyon, 2023-03-28:
Hi, you can already connect with MQTT explorer and trigger a reboot by setting the maintenance/reset node to 1.

bcutter, 2023-03-28:
OK that's good news - and a welcome prerequisite for the target of having an entity for that in HA.

Step 1: how to template this? (MQTT noob asking)

Step 2: why not making this button available out of the box?

desheh, 2023-03-31:
You can easily setup a button (trigger the "MQTT: Publish" service) that will publish an MQTT packet with topic: `nuki/maintenance/reset` with payload `1`.

bcutter, 2023-03-31:
> You can easily setup a button (trigger the "MQTT: Publish" service) that will publish an MQTT packet with topic: `nuki/maintenance/reset` with payload `1`.

How?

To be precise: not talking bout a simple GUI button but a `button.nuki_hub_restart` **entity**.

If one could provide a full working example, that would be helpful I guess.

benedikt-bartscher, 2023-03-31:
You can either template the button with Home Assistant (https://www.home-assistant.io/integrations/template/) manually or extend nuki_hub by adding mqtt autodiscovery for the button

bcutter, 2023-03-31:
Sorry, still no news - I'm really looking for a _working example_ on how to achieve this.

mundschenk-at, 2023-04-02:
@bcutter I don't want to sound harsh, but you could look that up in HA's [MQTT documentation](https://www.home-assistant.io/integrations/button.mqtt/).

technyon, 2023-04-04:
Check Release 8.21.

closed 2023-04-04 by technyon

bcutter, 2023-04-04:
[image]

Veeeeery nice. Two new entites per NUKI hub! 🎉🥳

And the restart works like a charme. Thank you so much!