technyon/nuki_hub #160: Feature request: Create update entity for NUKI Hub in Home Assistant

bcutter, 2023-03-28:
As we now have the `sensor.door_a_nuki_hub_version` entity since one of the last releases, one can use the GitHub integration, get the latest tag entity from https://github.com/technyon/nuki_hub and compare both with a template sensor to create a custom "NUKI hub firmware update available" sensor.

Nowadays Home Assistant uses update entities which can be provided by integrations. I'm not sure if NUKI Hub is able to provide those (that's actually different from that autodiscovery magic from MQTT, right? - as NUKI hub is not an integration). It would be great to have a (similar) convenient "out of the box" solution in Home Assistant for new NUKI Hub updates.

technyon, 2023-03-29:
No, there's no easy way to provide this from NUKI Hub side. Updates are on Github, and querying that is outside the scope of this project. Maybe something can be done on home assitant side (I actually read there's a github integration somewhere).

mundschenk-at, 2023-03-29:
You can't create `update` entities via the `template` platform in Home Assistant but you can with `mqtt`. I think it should be possible to create an automation blueprint that checks GitHub and creates the necessary MQTT topics for a proper 'update` entity. Installation will not be totally straight-forward though (unless Nuki Hub would add the auto-discovery payload for that that update entity, which may be too specific to be generally useful.)

bcutter, 2023-03-29:
I've implemented this on my own already, I get a notification from my HA instance once there's a newer NUKI Hub version for at least one of my NUKI Hubs. A bit of visualization in the GUI on top, done.

Input:
- GitHub latest tag sensor
- NUKI Hub version sensor

Output:
- Template sensor "NUKI Hub update available" matching those two input sensors
- GUI visualization
- Notification automation

I was aiming at automating things as far as possible (next step: actually DO an update automatically by downloading the .7z, extracting the needed .bin and magically POSTing it to the /ota page). An out of the box update entity would have been nice (but necessarily needed, a simple template entity is working just fine too).

So as there's understandable no low hanging fruit her, I'm fine with closing this as not planned because not possible. Feel free.

closed 2023-03-30 by technyon

bcutter, 2024-07-22:
Any possibility to implement this nowadays? :-)

bcutter, 2024-07-23:
> home assistant update sensor

Great, that's exactly what I was looking (and asking) for in this feature request. So the 9.00 series is still a dev branch I guess? Any estimations? 8.xx seems to be quite rock solid so far so I don't expect a point 9 release to arrive shortly.

iranl, 2024-07-23:
9.00 will be the next release because of the move to Arduino core 3 and IDF 5.1.

There are some BLE issues that are reported on Discord still being looked at but other than that it is ready.

There already is a beta available.