technyon/nuki_hub #162: Failed update of lock battery percentage

amastrogiacomo1968, 2023-03-31:
Good morning, I have noticed that the battery percentage update is not happening, let me explain better. The percentage has remained fixed at 68% for weeks, yesterday I tried unplugging and re-plugging the battery and it dropped to 58%. Today I tried again and it went down to 54%. How can I solve this problem?
Thank you.

technyon, 2023-03-31:
- Post the content of the system information page
- Please get logs (either MQTT or serial)

amastrogiacomo1968, 2023-04-01:
Log mqtt

2023-03-30 06:53:29: Saving in-memory database to /data//mosquitto.db.
2023-03-30 07:23:30: Saving in-memory database to /data//mosquitto.db.
2023-03-30 07:53:31: Saving in-memory database to /data//mosquitto.db.
2023-03-30 08:23:32: Saving in-memory database to /data//mosquitto.db.
2023-03-30 08:53:33: Saving in-memory database to /data//mosquitto.db.
2023-03-30 09:23:34: Saving in-memory database to /data//mosquitto.db.
2023-03-30 09:53:35: Saving in-memory database to /data//mosquitto.db.
2023-03-30 10:23:36: Saving in-memory database to /data//mosquitto.db.
2023-03-30 10:53:37: Saving in-memory database to /data//mosquitto.db.
2023-03-30 10:59:23: Client 0iuKSelz40dJnVejhzzMwP closed its connection.
2023-03-30 10:59:51: New connection from 192.168.1.18:41033 on port 1883.
2023-03-30 10:59:51: New client connected from 192.168.1.18:41033 as 4XFT9mGJBOp1c3BnzWZrlY (p2, c1, k60, u'mqtt').
2023-03-30 11:23:38: Saving in-memory database to /data//mosquitto.db.
2023-03-30 11:53:39: Saving in-memory database to /data//mosquitto.db.
2023-03-30 12:23:40: Saving in-memory database to /data//mosquitto.db.
2023-03-30 12:53:41: Saving in-memory database to /data//mosquitto.db.
2023-03-30 13:23:42: Saving in-memory database to /data//mosquitto.db.
2023-03-30 13:53:43: Saving in-memory database to /data//mosquitto.db.
2023-03-30 13:55:16: New connection from 192.168.1.234:60502 on port 1883.
2023-03-30 13:55:16: Client nukihub already connected, closing old connection.
2023-03-30 13:55:16: New client connected from 192.168.1.234:60502 as nukihub (p2, c0, k15, u'mqtt').
2023-03-30 14:23:44: Saving in-memory database to /data//mosquitto.db.
2023-03-30 14:53:45: Saving in-memory database to /data//mosquitto.db.
2023-03-30 15:23:46: Saving in-memory database to /data//mosquitto.db.
2023-03-30 15:53:47: Saving in-memory database to /data//mosquitto.db.
2023-03-30 16:23:48: Saving in-memory database to /data//mosquitto.db.
2023-03-30 16:53:49: Saving in-memory database to /data//mosquitto.db.
2023-03-30 17:23:50: Saving in-memory database to /data//mosquitto.db.
2023-03-30 17:53:51: Saving in-memory database to /data//mosquitto.db.
2023-03-30 18:23:52: Saving in-memory database to /data//mosquitto.db.
2023-03-30 18:53:53: Saving in-memory database to /data//mosquitto.db.
2023-03-30 19:23:54: Saving in-memory database to /data//mosquitto.db.
2023-03-30 19:53:55: Saving in-memory database to /data//mosquitto.db.
2023-03-30 20:23:56: Saving in-memory database to /data//mosquitto.db.
2023-03-30 20:53:57: Saving in-memory database to /data//mosquitto.db.
2023-03-30 21:23:58: Saving in-memory database to /data//mosquitto.db.
2023-03-30 21:53:59: Saving in-memory database to /data//mosquitto.db.
2023-03-30 22:24:00: Saving in-memory database to /data//mosquitto.db.
2023-03-30 22:47:00: Client 4XFT9mGJBOp1c3BnzWZrlY closed its connection.
2023-03-30 22:47:26: New connection from 192.168.1.18:38121 on port 1883.
2023-03-30 22:47:26: New client connected from 192.168.1.18:38121 as 09fi9KKT0Fmh3j5VJ8iLCt (p2, c1, k60, u'mqtt').
2023-03-30 22:54:01: Saving in-memory database to /data//mosquitto.db.
2023-03-30 23:24:02: Saving in-memory database to /data//mosquitto.db.
2023-03-30 23:54:03: Saving in-memory database to /data//mosquitto.db.
2023-03-31 00:24:04: Saving in-memory database to /data//mosquitto.db.
2023-03-31 00:54:05: Saving in-memory database to /data//mosquitto.db.
2023-03-31 01:24:06: Saving in-memory database to /data//mosquitto.db.
2023-03-31 01:54:07: Saving in-memory database to /data//mosquitto.db.
2023-03-31 02:24:08: Saving in-memory database to /data//mosquitto.db.
2023-03-31 02:54:09: Saving in-memory database to /data//mosquitto.db.
2023-03-31 03:24:10: Saving in-memory database to /data//mosquitto.db.
2023-03-31 03:54:11: Saving in-memory database to /data//mosquitto.db.
2023-03-31 04:24:12: Saving in-memory database to /data//mosquitto.db.
2023-03-31 04:54:13: Saving in-memory database to /data//mosquitto.db.
2023-03-31 05:24:14: Saving in-memory database to /data//mosquitto.db.
2023-03-31 05:54:15: Saving in-memory database to /data//mosquitto.db.
2023-03-31 06:24:16: Saving in-memory database to /data//mosquitto.db.
2023-03-31 06:54:17: Saving in-memory database to /data//mosquitto.db.
2023-03-31 07:24:18: Saving in-memory database to /data//mosquitto.db.
2023-03-31 07:54:19: Saving in-memory database to /data//mosquitto.db.
2023-03-31 08:24:20: Saving in-memory database to /data//mosquitto.db.
2023-03-31 08:54:21: Saving in-memory database to /data//mosquitto.db.
2023-03-31 09:24:22: Saving in-memory database to /data//mosquitto.db.
2023-03-31 09:54:23: Saving in-memory database to /data//mosquitto.db.
2023-03-31 10:24:24: Saving in-memory database to /data//mosquitto.db.
2023-03-31 10:54:25: Saving in-memory database to /data//mosquitto.db.
2023-03-31 11:24:26: Saving in-memory database to /data//mosquitto.db.
2023-03-31 11:54:27: Saving in-memory database to /data//mosquitto.db.
[... 7 lines cut ...]
2023-03-31 13:38:39: New connection from 192.168.1.18:46953 on port 1883.
2023-03-31 13:38:39: New client connected from 192.168.1.18:46953 as 7KITKgEWH0lvGtejlCW4TU (p2, c1, k60, u'mqtt').
2023-03-31 13:54:31: Saving in-memory database to /data//mosquitto.db.
2023-03-31 14:24:32: Saving in-memory database to /data//mosquitto.db.
2023-03-31 14:54:33: Saving in-memory database to /data//mosquitto.db.
2023-03-31 15:24:34: Saving in-memory database to /data//mosquitto.db.
[... cut ...]
2023-03-31 21:24:46: Saving in-memory database to /data//mosquitto.db.
2023-03-31 21:54:47: Saving in-memory database to /data//mosquitto.db.
2023-03-31 22:24:48: Saving in-memory database to /data//mosquitto.db.
2023-03-31 22:54:49: Saving in-memory database to /data//mosquitto.db.
2023-03-31 23:24:50: Saving in-memory database to /data//mosquitto.db.
2023-03-31 23:54:51: Saving in-memory database to /data//mosquitto.db.
2023-04-01 00:24:52: Saving in-memory database to /data//mosquitto.db.
2023-04-01 00:54:53: Saving in-memory database to /data//mosquitto.db.

mundschenk-at, 2023-04-01:
I think @technyon meant the ESP's logs (which you can get by either attaching to the serial port or enabling MQTT logging to `<prefix>/maintenance/log`.

amastrogiacomo1968, 2023-04-02:
This?
Porta d'ingresso (lock.porta_d_ingresso)
MQTT discovery data:
Topic: homeassistant/lock/2856dffd/smartlock/config
Payload
device:
  identifiers:
    - nuki_2856dffd
  manufacturer: Nuki
  model: SmartLock
  name: Porta d'ingresso
name: Porta d'ingresso
unique_id: 2856dffd_lock
command_topic: nuki/lock/action
payload_lock: lock
payload_unlock: unlock
payload_open: unlatch
state_topic: nuki/lock/binaryState
state_locked: locked
state_unlocked: unlocked
optimistic: 'false'
platform: mqtt
Subscribed topics:
nuki/lock/binaryState
2 messaggi ricevuti più di recente
Received 11:58:31
QoS: 0, Retained
Payload: unlocked
Received 13:05:42
QoS: 0
Payload: unlocked
Transmitted messages:
nuki/lock/action
1 messaggi trasmessi più di recente
Transmitted 11:59:19
QoS: 0
Payload: unlock
Porta d'ingresso battery low (binary_sensor.porta_d_ingresso_battery_low)
MQTT discovery data:
Topic: homeassistant/binary_sensor/2856dffd/battery_low/config
Payload
device:
  identifiers:
    - nuki_2856dffd
  manufacturer: Nuki
  model: SmartLock
  name: Porta d'ingresso
name: Porta d'ingresso battery low
unique_id: 2856dffd_battery_low
device_class: battery
state_topic: nuki/battery/critical
entity_category: diagnostic
payload_on: '1'
payload_off: '0'
platform: mqtt
Subscribed topics:
nuki/battery/critical
2 messaggi ricevuti più di recente
Received 11:58:35
QoS: 0, Retained
Payload: 0
Received 13:05:42
QoS: 0
Payload: 0
Transmitted messages:
Porta d'ingresso battery voltage (sensor.porta_d_ingresso_battery_voltage)
MQTT discovery data:
Topic: homeassistant/sensor/2856dffd/battery_voltage/config
Payload
device:
[... 420 lines cut ...]
device:
  identifiers:
    - nuki_2856dffd
  manufacturer: Nuki
  model: SmartLock
  name: Porta d'ingresso
name: Porta d'ingresso NUKI Hub version
unique_id: 2856dffd_nuki_hub__version
state_topic: nuki/info/nukiHubVersion
entity_category: diagnostic
enabled_by_default: 'true'
icon: mdi:counter
platform: mqtt
Subscribed topics:
nuki/info/nukiHubVersion
2 messaggi ricevuti più di recente
Received 11:58:35
QoS: 0, Retained
Payload: 8.20
Received 13:05:37
QoS: 0
Payload: 8.20
Transmitted messages:
Attivazioni
Nessuna attivazione

mundschenk-at, 2023-04-02:
Sorry, no. You need to enable MQTT logging in your Nuki Hub configuration and at something like this to your HA configuration. Alternatively, you can attach the ESP to a computer and read the serial logs from the USB connection.

```yaml
mqtt:
  sensor:
    - name: "Nuki Hub Log"
      state_topic: "<prefix>/maintenance/log"
```

amastrogiacomo1968, 2023-04-03:
I'm sorry but I'm poorly prepared on this topic :-(
You mean this log:
Nuki Hub Log cambiato in Reading config. Result: success
22:18:20 - 11 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
22:18:20 - 11 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
22:18:19 - 11 ore fa
Nuki Hub Log cambiato in HASS setup for lock completed.
22:03:22 - 11 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
22:03:21 - 11 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
22:03:21 - 11 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
22:03:21 - 11 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
22:03:21 - 11 ore fa
Nuki Hub Log cambiato in Nuki paired
22:03:17 - 11 ore fa
Nuki Hub Log cambiato in Nuki start pairing
22:03:17 - 11 ore fa
Nuki Hub Log cambiato in MQTT connected
22:03:14 - 11 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
21:50:41 - 11 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
21:50:41 - 11 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
21:50:41 - 11 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
21:48:54 - 11 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
21:35:42 - 12 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
21:35:42 - 12 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
21:35:42 - 12 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
21:33:57 - 12 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
21:20:43 - 12 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
21:20:42 - 12 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
21:20:42 - 12 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
21:18:54 - 12 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
21:05:42 - 12 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
21:05:42 - 12 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
21:05:42 - 12 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
21:03:57 - 12 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
20:50:41 - 12 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
20:50:41 - 12 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
20:50:41 - 12 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
20:48:56 - 12 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
20:35:42 - 13 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
20:35:42 - 13 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
20:35:42 - 13 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
20:33:25 - 13 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
20:20:42 - 13 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
20:20:42 - 13 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
20:20:42 - 13 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
20:18:37 - 13 ore fa
Nuki Hub Log cambiato in Reading advanced config. Result: success
20:05:42 - 13 ore fa
Nuki Hub Log cambiato in Reading config. Result: success
20:05:42 - 13 ore fa
Nuki Hub Log cambiato in Querying lock battery state: success
20:05:42 - 13 ore fa
Nuki Hub Log cambiato in Querying lock state: unlocked
20:03:36 - 13 ore fa
Nuki Hub Log cambiato in Lock action result: success
20:03:34 - 13 ore fa
Nuki Hub Log cambiato in Lock action received: unlock
20:03:30 - 13 ore fa

technyon, 2023-04-03:
Querying the battery state works fine:

`Nuki Hub Log cambiato in Querying lock battery state: success`

It's hard to tell what's going wrong, but it must be something in your setup. I'd suggest to download MQTT Explorer and check the nodes within the MQTT tree: "battery/level" should show the battery level in percent. Also check the discovery topic under "homeassistant/sensor/<device id>/battery_level.

amastrogiacomo1968, 2023-04-04:
Thanks, that's really weird. In my ignorance on the subject I saw that the value was sent punctually.
I will investigate the investigation as instructed and update you.
Out of curiosity, when does the battery percentage drop over time?

amastrogiacomo1968, 2023-04-04:
"For example, how long does it take to go from 54% to 52%?"

technyon, 2023-04-04:
It depends on the battery you use and the amount of lock operations per day, and also how much battery it uses per lock operation, so I don't really have an answer to that. I use a percent or two a day with a Power Pack, but I've permanently connected by power pack to a charger, so it doesn't matter for me.

amastrogiacomo1968, 2023-04-04:
Of course it depends on how you use it I was asking to regulate myself, a bill that drops by 1% in a week a bill that drops by 5% in a day

technyon, 2023-04-08:
So, is this issue solved for you now?

I can't give you an exact number what to expect, what I can say with the power pack for me after charging to 100% it doesn't drop at all for a few days. Then it looses a percent every day or so, but this could be way lower if using non-rechargable batteries.

amastrogiacomo1968, 2023-04-08:
Unfortunately the value has remained fixed at 54% for about a week, I would say that it is not resolved but I don't even know if it can be resolved (even if it worked perfectly in the first versions of the sw). Thanks anyway for the support

technyon, 2023-04-08:
Does the smartphone app show a different value?

amastrogiacomo1968, 2023-04-08:
Unfortunately I don't use the app. I tried taking the battery pack off and on and the value dropped from 54% to 50%, I don't know if that gives you any more information

mundschenk-at, 2023-04-08:
@amastrogiacomo1968 It's possible that the estimated percentage does not change that much per week ad mid-charge levels. My battery percentage also has not changed for some time now, but its the same in the app. (Alternatively, there might be a bug in the most recent lock firmware.)

amastrogiacomo1968, 2023-04-08:
So I'd better reconnect the original bridge and the app and compare the values

mundschenk-at, 2023-04-08:
No need for the bridge. Just start the app within BLE range.

amastrogiacomo1968, 2023-04-08:
App installed and configured, the battery percentage is now the same between the Nuki app and the value read in the Home Assistant. I will monitor in the coming weeks and update you. I would say not to close the issue for now

amastrogiacomo1968, 2023-04-19:
The percentage updates regularly, I close and thank you for the support

closed 2023-04-19 by amastrogiacomo1968

technyon, 2023-04-19:
Glad to hear