technyon/nuki_hub #163: in home assistant the lock status of jammed cannot be shown?

bertybassett, 2023-04-01:
what am I doing wrong?

technyon, 2023-04-02:
The status only shows locked or unlocked. You can however look into the MQTT tree to see the actual status (lock/state).

bertybassett, 2023-04-02:
```
mqtt:
  sensor:
    - name: "nuki_motorstatus"
      unique_id: '12bAAA3asde'
      state_topic: "nuki/lock/state"
```
Was the answer shame the auto discovery on home assistant does not include this information

technyon, 2023-04-02:
As far as I understand, this is a HA limitation, as the smartlock device class doesn't have a jammed state (correct me if I'm wrong),

mundschenk-at, 2023-04-02:
It does: 
[image]

bertybassett, 2023-04-02:
https://www.home-assistant.io/integrations/lock.mqtt/

An MQTT lock can also report the intermediate states unlocking, locking or jammed if the motor reports a jammed state.

Just saying there is a jammed status in the MQTT lock within Home Assistant, don't worry too much as I fixed my issue using the above.

bertybassett, 2023-04-02:
looks like my fudge work around never worked.

leranp, 2023-05-12:
Same here, sometimes i have jammed situations but there's no indication about this in the sensors.
What can i do ? Do i need to create a manually mqtt sensor?

PR #296 (2024-02-18): Enable Additional Lock states in Home Assistant + Make Opener Continuous mode an option for lock/unlock
**Allow Home Assistant to show intermediate states (unlocking/locking/opening) and jammed state**

Note: Not all intermediate states are captured by the Hub and these will not show up in Home Assistant. This change will just make Home Assistant more accurately reflect the state of nuki/lock/state instead of only being able to show locked/unlocked.

Closes #253, Closes #163

**Add a separate MQTT topic for continuous mode state, distinct from lock state
Make Opener Continuous mode (instead of [... cut]
closed 2024-02-18 by technyon