technyon/nuki_hub #164: nuki hub and domoticz

Epik77, 2023-04-01:
HI,
I'm using this fantastic hub interfaced with domoticz
using Home Assistant discovery topic and close button LED and button are detected automatically.
Everything works fine.
Every time I restart the domoticz hub it gives me this MQTT error:

  2023-04-01 18:09:47.819 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/trigger/config/message: {"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance door"},"~":"nuki","name":"trigger entrance door","unique_id":"XXXXXXXX_trigger","stat_t":"~/lock/trigger","ent_cat":" diagnostic","enabled_by_default":"true"}
2023-04-01 18:09:47.920 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/firmware_version/config/message: {"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance door"},"~":"nuki","name":"entrance door Firmware version","unique_id":"XXXXXXXX_firmware_version","stat_t":"~/info/firmwareVersion","ent_cat": "diagnostic","enabled_by_default":"true","ic":"mdi:counter"}
2023-04-01 18:09:48.020 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/hardware_version/config/message: {"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance door"},"~":"nuki","name":"entrance door Hardware version","unique_id":"XXXXXXXX_hardware_version","stat_t":"~/info/hardwareVersion","ent_cat": "diagnostic","enabled_by_default":"true","ic":"mdi:counter"}
2023-04-01 18:09:48.121 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/nuki_hub_version/config/message: {"dev":{"ids":["nuki_34e7a35d"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance port"},"~":"nuki","name":"entrance port NUKI Hub version","unique_id":"XXXXXXXX_nuki_hub__version","stat_t":"~/info/nukiHubVersion","ent_cat" :"diagnostic","enabled_by_default":"true","ic":"mdi:counter"}
2023-04-01 18:09:49.537 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/trigger/config/message: {"dev":{"ids":["nuki_34e7a35d"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance door"},"~":"nuki","name":"trigger entrance door","unique_id":"XXXXXXXX_trigger","stat_t":"~/lock/trigger","ent_cat":" diagnostic","enabled_by_default":"true"}
2023-04-01 18:09:49.639 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/firmware_version/config/message: {"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance door"},"~":"nuki","name":"entrance door Firmware version","unique_id":"34e7a35d_firmware_version","stat_t":"~/info/firmwareVersion","ent_cat": "diagnostic","enabled_by_default":"true","ic":"mdi:counter"}
2023-04-01 18:09:49.740 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/hardware_version/config/message: {"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance door"},"~":"nuki","name":"entrance door Hardware version","unique_id":"XXXXXXXX_hardware_version","stat_t":"~/info/hardwareVersion","ent_cat": "diagnostic","enabled_by_default":"true","ic":"mdi:counter"}
2023-04-01 18:09:49.841 Error: MQTT Homeassistant: MQTT_Discovery (on_auto_discovery_message): Error: Value is not convertible to bool.! (topic: homeassistant/sensor/XXXXXXXX/nuki_hub_version/config/message: {"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name" :"entrance port"},"~":"nuki","name":"entrance port NUKI Hub version","unique_id":"XXXXXXXX_nuki_hub__version","stat_t":"~/info/nukiHubVersion","ent_cat" :"diagnostic","enabled_by_default":"true","ic":"mdi:counter"}

If it is not possible to fix this, I can still disable Home Assistant discovery and create a plugin that uses mqtt but it would be a shame not to make it compatible with other platforms.

Thank you all for the excellent work you have done!

mundschenk-at, 2023-04-01:
Is that strike-through text part of the error message or did your log get mangled? Please mark it up as a code block.

Epik77, 2023-04-01:
sorry :-)
I replaced the device ID code with XXXXX

mundschenk-at, 2023-04-01:
@technyon `enabled_by_default` is a boolean, so `"true"` should be `true` ( i.e. without the quotes). Apparently the JSON parser in Home Assistant is more lenient.

technyon, 2023-04-02:
Try this binary:

[nuki_hub-8.21-pre-1.zip](https://github.com/technyon/nuki_hub/files/11131586/nuki_hub-8.21-pre-1.zip)

Epik77, 2023-04-02:
Excellent with this firmware when restarting I have no errors. It remains only to fix the battery level of some devices
[image]
thanks for the support :-)

technyon, 2023-04-04:
Check Release 8.21.

closed 2023-04-04 by technyon