technyon/nuki_hub #165: Jamms out of sync with app

bertybassett, 2023-04-02:
I've noticed when the nuki app and HomeKit detect a jammed 10 seconds later they both displays as locked

Now the ESP32 bridge when it detects a jam it stays as unlocked in HA which causes the system to get out of sync.

Any options in the frontend to stay trick a jam like the nuki app/homekit on/off would be great.

technyon, 2023-04-02:
How does it go out of sync, I'm not sure sure what you mean by that. I'm not a big of creating such artifical states if not absolutely necessary, it's a guessing game.

Why does your lock get jammed in the first place? That did happen to me too, but it was due to a lack of lubricant. Easily fixed by addind some lubricant to the lock, maybe it's better to fix the mechanical problem than having NUKI Hub creating workarounds for such rare cases.

bertybassett, 2023-04-02:
Us European cousin (I am in the UK) have to lift our handles to look the door.  If I come in and forget to lift the handle the lock jams.  Once jammed it becomes out of sync, in HA it will say unlocked however nuki say locked.

technyon, 2023-04-02:
I think that's some kind of bug. Can you use MQTT explorer to check both lock/state and lock/binaryState (the latter is used by HA)

bertybassett, 2023-04-02:
It’s because the nuki app shows jammed for 10 seconds then resets to locked whereas your app stays on unlocked and gets out of sync.

technyon, 2023-04-02:
I understand the issue, but I can't do anything about it unless I know what's going on. Please provide more information:

- Post the system information page
- Download MQTT explorer and check the nodes I mentioned before. If binaryState doesn't change back to the correct state, there must be some bug, but I want to confirm this.
- Do you use a NUKI Bridge alongside NUKI Hub?

technyon, 2023-04-08:
@bertybassett Any update on this, were you able to try some of my suggestions?

bertybassett, 2023-04-08:
MQTT lock shows as unlocked

If I don't lift handle to make it jam.

Then press lock 

MQTT show unlocked and motor show jammed

vs nuki app which shows jammed for 10 seconds then locked.

Basically your logic is wrong for jams compared to nuki app.

technyon, 2023-04-08:
One question you haven't answered: Do you use a NUKI Bridge together with NUKI Hub?

I've never had a jammed motor, so I can't say too much about this. What do you do when motor jammed?

I'm not sure about the logic, maybe when the app is open it polls the lock, while NUKI Hub has to wait for a notification from the lock that the state has changed. My guess is the lock doesn't send this notification unless you complete another lock operation

bertybassett, 2023-04-08:
no hub

bertybassett, 2023-04-08:
just HomeKit and nuki app.

bertybassett, 2023-04-08:
got it working in Home Assistant with 2 buttons -  unlock +  lock  and a status of motor entity locked/unlocked/Jammed.  this kind of works.

technyon, 2023-04-08:
I have a feeling your problem is that you're using homekit: If I remember correctly, if you enable homekit integration, the notification about state changes is different. Just for testing, maybe try to disable homekit in the smartphone app.

technyon, 2023-04-22:
Closing due to inactivity.

closed 2023-04-22 by technyon