technyon/nuki_hub #166: Status of the device

leranp, 2023-04-05:
Hi,

Thank you for your project!
Recently my ESP device was disconnected from the network, but the automation in HA didn't sent me a message about it, so i tried to see why and saw that the device in HA isn't going to "unavailable" when it's going offline, Do I miss something?
It looks like there is no "Offline/Online" MQTT sensor that switches the device to unavailable state

technyon, 2023-04-05:
Which firmware version are you on?

mundschenk-at, 2023-04-05:
@technyon That's something for the auto-discovery rewrite.

technyon, 2023-04-05:
Well yes, but the latest firmware has a connection state topic (including auto discovery) which does exactly this.

mundschenk-at, 2023-04-05:
No, not in the sense that the OP likely means. In HA, when a device becomes unavailable, the state of the entities changes to `unavailable` (or `unknown` with slightly different semantics). To get that behavior with MQTT, the discovery payload has to include some additional keys.

technyon, 2023-04-05:
ok I see

yasars, 2023-05-05:
same problem here, yes, I also have after a restart that sometimes the status is unavailable ... after a renewed restart it works again ... I'll try with the automation if I can do something that it reboots again if unavailable ... so it would be on small fixx....

mundschenk-at, 2023-05-05:
> same problem here, yes, I also have after a restart that sometimes the status is unavailable ... after a renewed restart it works again ... I'll try with the automation if I can do something that it reboots again if unavailable ... so it would be on small fixx....

Please open your own issue, this is the opposite of what the OP wants (explicitly setting the state to `unavailable` because currently it isn't).

iranl, 2024-02-07:
After merging #292 all entities in Home Assistant discovery now include the activity topic which (in conjunction with LWT) ensures all entities (not just the lock entity) become unavailable when the device goes offline/becomes unavailable to the MQTT broker for whatever reason. Tested in my own build including #292 by pulling the USB power cable from the wall.

I propose this issue is marked as fixed by #292 and closed.

technyon, 2024-02-08:
ok, closing this

closed 2024-02-08 by technyon