technyon/nuki_hub #167: Configuration export/backup

Arn0uDz, 2023-04-07:
Would love the ability to export the current configuration in case of hardware failure of the esp32 device.

The configuration is pretty basic, but new features keep getting added pretty quickly.

technyon, 2023-04-08:
Although this would be a nice feature, it's unlikely to happen. The API of "Preferences" used for storing all the setting is nice to use, but not very well suited for backups. I'd need to manually serialize every single setting and then deserialize on restore which is error-prone and a lot work.

bcutter, 2023-04-08:
Only workaround:

Turn to status page and copy everything to a local text file. On another ESP set everything accordingly.

Not really a backup, but the current option and way to go for migration.

technyon, 2023-04-08:
Not ideal I agree, but a hardware failure on an ESP is rare.

Arn0uDz, 2023-04-08:
Doesn't have to be a hardware failure, could also be software. For instance the esp is not reachable anymore and requires a wipe and reflash, thus losing all the settings. 

For the moment the easiest way is to take screenshots of my settings.

closed 2023-05-06 by technyon