technyon/nuki_hub #168: Unable to use OTA update

Mike38saw, 2023-04-07:
Hi,

The latest versio that i installed OTA was the 8.11. From this, i'm not able to use the OTA updater with this message.

OTA functionality not ready. Please wait a moment and reload.

Any suggestions ?

Thanks,

technyon, 2023-04-07:
I guess you did wait a moment (60 secs to be exact)?

The only explanation would be a reboot loop, can you get serial logs?

Mike38saw, 2023-04-08:
Hi, i'm sorry, how to retrieve these logs ?

technyon, 2023-04-08:
Download hterm:

https://www.der-hammer.info/pages/terminal.html

Attach the ESP to your computer, then connect via hterm to the ESP. Change "Newline at" to "CT+LF" to make the output readable.

Mike38saw, 2023-04-08:
Restart timer expired, restarting device.
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (519) esp_core_dump_flaKr½core dump partition found!
E (519) esp_core_dump_flash: No core dump partition found!
NUKI Hub version 8.11
Hardware detect     : 2
Hardware detect GPIO: 26
Using PIN 26 for network device selection
Network device: Builtin WiFi
MQTT without TLS.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: Livebox-1AC0
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.1.43
WiFi connected: 192.168.1.43
Host name: nukihub
MQTT Broker: 192.168.1.251:1883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: m.s
MQTT connected
Nuki start pairing
Nuki paired
Querying lock state: lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Querying lock battery state: failed
Reading config. Result: failed
Reading advanced config. Result: failed
Querying lock state: Querying lock state: No BLE beacon received from the lock for 18 seconds, restarting device.
Restart timer expired, restarting device.
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (519) esp_core_dump_flaKr½core dump partition found!
E (519) esp_core_dump_flash: No core dump partition found!
NUKI Hub version 8.11
Hardware detect     : 2
Hardware detect GPIO: 26
Using PIN 26 for network device selection
Network device: Builtin WiFi
MQTT without TLS.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: Livebox-1AC0
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.1.43
WiFi connected: 192.168.1.43
Host name: nukihub
MQTT Broker: 192.168.1.251:1883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
[... 182 lines cut ...]
Hardware detect GPIO: 26
Using PIN 26 for network device selection
Network device: Builtin WiFi
MQTT without TLS.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: Livebox-1AC0
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.1.43
WiFi connected: 192.168.1.43
Host name: nukihub
MQTT Broker: 192.168.1.251:1883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: m.s
MQTT connected
Nuki start pairing
Nuki paired
Querying lock state: Querying lock battery state: failed
Reading config. Result: failed
Reading advanced config. Result: failed
Querying lock state: Querying lock state:

technyon, 2023-04-08:
I think you've set the "Restart timer" option to a low value. This will reboot your ESP unconditionally after the configured a time. Enabling this is usually unnecessary, you should set it to -1 to disable it.

Mike38saw, 2023-04-08:
Thanks for this answer. Is it not requiered to use this option if i want to restart the hub in case of BLE or other errors ? Thanks

mundschenk-at, 2023-04-08:
No, it is not.

technyon, 2023-04-08:
If you want to restart in case of ble errors, set "Restart if bluetooth beacons not received" (60 seconds should be fine).

To restart on connection errors, set "Network Timeout until restart".

Mike38saw, 2023-04-08:
Alright, it has solved the issue. I also put the Network timeout until restart to 60.

I was able to make the OTA Update. Thank you so mich for your help

closed 2023-04-08 by technyon