technyon/nuki_hub #171: Nuki lock state: undefined

Rebislori, 2023-04-12:
Ho everyone.
For the last month, the nukihub works only id i switch off the internal wifi of my smart lock 3.0 pro.

When it's switched off everything works Great and The hub is very responsive.

If i switch internal wifi on, it just stops working!
The status is always on "undefined", and no changes are registered.
Am i missing some information?

[image]

Rebislori, 2023-04-12:
If i turn on wifi, nukihub stops working. Or better, it stays like in the picture i poste a ove.

With wifi turned off, everything works extremely well (by the way, thanks for your job!).

I just wanted to know if it's a known issue, or if i miss something in the documentation.

technyon, 2023-04-12:
You didn't miss anything, and it's not the first report wifi disables bluetooth functionality (which is I think what happens). Not all users with a Pro lock can confirm this though, I can't say for sure what's going on or what sense it makes to disable blueooth. I guess you'll have to make a choice, NUKI Hub or NUKI Cloud.

micudaj, 2023-04-28:
I have the same issue but even after disabling the wifi of the nuki 3.0 pro I still get the undefined state. Any idea what the problem might be? I paired it as an app as I have a bridge I also connected to the lock

mundschenk-at, 2023-05-01:
> I have the same issue but even after disabling the wifi of the nuki 3.0 pro I still get the undefined state. Any idea what the problem might be? I paired it as an app as I have a bridge I also connected to the lock

When paired as an app, NukiHub does not receive the necessary updates from the BLE API. This is by design (on Nuki's side).

mundschenk-at, 2023-05-01:
@technyon I think this issue can be closed as a duplicate of #141.

technyon, 2023-05-01:
Correct.

@Rebislori Due to NUKI, it's not possible to use Wifi and Bridge functionality at the same time, this is by design.

closed 2023-05-01 by technyon