technyon/nuki_hub #173: No Nuki in Home Assistant

alkertrazz, 2023-04-13:
Hey guys,

first of all thank you for all of your work!

Unfortunately it isn't possible for home assistant to discover my Nuki 3.0. MQTT is connected, the lock is paired but everything what I try fails.

flotomation, 2023-04-13:
Could you check that you configured the auto config topic?

[image]

alkertrazz, 2023-04-17:
Hey flotomation, 
thx this were the problem. I didn't read all of the settings because I thought this advanced settings were only for SSL.

closed 2023-04-19 by technyon

platini76, 2023-07-14:
I have the same problem, all is configured. mqtt is connected... discovery topic is configured.. but ... no entity found, can someone help me?

craigrouse, 2023-07-17:
I have the same issue as @platini76, and can't figure out why autodiscovery isn't working