technyon/nuki_hub #174: Can't connect to mqtt broker

burakcan, 2023-04-14:
Hello, I am trying to install nuki_hub on a device. It seems like everything works but I couldn't connect it to a mqtt broker running on EMQX (hosted). I also tried with other hosted services but it was the same. But it works with public instances (no authorization/authentication) Not sure what I am doing wrong. Can you give me some pointers? Here is the output from the device:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (527) esp_core_dump_fl����� No core dump partition found!
E (527) esp_core_dump_flash: No core dump partition found!
NUKI Hub version 8.22
IP address empty, falling back to DHCP.
IP configuration: DHCP
Hardware detect     : 1
Network device: Wifi only
MQTT without TLS.
*wm:AutoConnect 
*wm:Connecting to SAVED AP: hallo
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.178.84
WiFi connected: 192.168.178.84
Host name: nukihub
MQTT Broker: [redacted].ala.us-east-1.emqxsl.com:8883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: nukihub
MQTT disconnected. Reason: TCP_DISCONNECTED
MQTT connect failed, rc=Free Heap: 118748
Nuki start pairing
Nuki paired
Querying lock state: Attempting MQTT connection
MQTT: Connecting with user: nukihub
MQTT disconnected. Reason: TCP_DISCONNECTED
MQTT connect failed, rc=Free Heap: 116832
unlocked
Querying lock battery state: success
Reading config. Result: success
Reading advanced config. Result: success
Attempting MQTT connection
MQTT: Connecting with user: nukihub
MQTT disconnected. Reason: TCP_DISCONNECTED
MQTT connect failed, rc=Free Heap: 113180
Attempting MQTT connection
MQTT: Connecting with user: nukihub
MQTT disconnected. Reason: TCP_DISCONNECTED
MQTT connect failed, rc=Free Heap: 113216

.... continues like this
```

mundschenk-at, 2023-04-14:
It looks like you are using a port usually set up for TLS without an TLS configuration on your Nuki Hub.

burakcan, 2023-04-14:
Ah, yes. Thank you it's connecting now :)

closed 2023-04-14 by burakcan