technyon/nuki_hub #175: Nuki Opener not pairing: CRC Error

apfelflo89, 2023-04-15:
Hi, 

i have problems with pairing my Nuki Oper with several esp32 boards now, so I can say it´s not a hardware thing. 

I tried the debug version from issue https://github.com/technyon/nuki_hub/issues/39

Could You look into thins?
Thats what the log states:

[163800][D][NukiBle.cpp:229] onResult(): Found nuki in pairing state: Nuki_Opener_XXXXXX addr: 54:d2:72:XX:XX
Nuki opener start pairing
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
[164242][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying
[164424][D][NukiBle.cpp:1105] onConnect(): BLE connected
[165336][D][NukiBle.cpp:604] pairStateMachine(): ##################### REQUEST REMOTE PUBLIC KEY #########################
[165463][E][NukiUtils.cpp:91] crcValid(): CRC CHECK FAILED!
[165493][E][NukiUtils.cpp:91] crcValid(): CRC CHECK FAILED!
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[194244][D][NukiBle.cpp:1111] onDisconnect(): BLE disconnected
[195408][W][NukiBle.cpp:749] pairStateMachine(): Pairing timeout
[195458][D][NukiBle.cpp:107] pairNuki(): pairing result 2
Nuki opener start pairing
[204126][D][NukiBle.cpp:1105] onConnect(): BLE connected
[205098][D][NukiBle.cpp:604] pairStateMachine(): ##################### REQUEST REMOTE PUBLIC KEY #########################
[205121][E][NukiBle.cpp:1033] handleReturnMessage(): Error: 10 for command: 00:01
Guru Meditation Error: Core  0 panic'ed (Double exception).

technyon, 2023-04-16:
Please flash version 8.22 and post the content of your the system information page.

flotomation, 2023-04-16:
I'm on 8.22 again, that what the info page looks like:
[image]

And that what's the logs give me: 
[image]

apfelflo89, 2023-04-16:
That's my system information page:

NUKI Hub version: 8.22
run: true
deviceId: 3420818637
mqttbroker: 10.1.1.7
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: false
mqttpath: nuki
openerena: true
mqttoppath: nukiopener
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihubopener
nettmout: -1
restdisc: false
resttmr: -1
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Opener firmware version: 
Opener hardware version: 
Opener paired: No
Opener PIN set: -
Opener has keypad: No
Network device: Built-in Wifi
Uptime: 1 minutes
Heap: 85484
Stack watermarks: nw: 6168, nuki: 1768, pd: 248
Restart reason FW: ConfigurationUpdated
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

apfelflo89, 2023-04-16:
And I'm stuck with this in the logs:
A pairing mode Opener right in front of the ESP

[image]

technyon, 2023-04-16:
ok, thanks.

The debug build from the other thread is quite old, and we've made fixes to the pairing code in the meantime.

Can you flash this binary and get new logs?

[nuki_hub-8.23-pre-2-dbg.zip](https://github.com/technyon/nuki_hub/files/11241687/nuki_hub-8.23-pre-2-dbg.zip)

apfelflo89, 2023-04-16:
Ok flashed it and that are the logs now: Still refering to CRC Errors

[ 39386][D][NukiBle.cpp:254] onResult(): Found nuki in pairing state: Nuki_Opener_1B15B3A2 addr: NUKI_BLE
Nuki opener start pairing
[ 39514][E][NukiBle.cpp:635] retrieveCredentials(): Getting data from NVS issue
[ 39514][D][NukiBle.cpp:85] pairNuki(): Nuki in pairing mode found
[ 39518][D][NukiBle.cpp:132] connectBle(): connecting within: nuki
[ 39523][D][NukiBle.cpp:139] connectBle(): connection attemnpt 0
[ 39608][D][NukiBle.cpp:1173] onConnect(): BLE connected
[ 40062][D][NukiBle.cpp:927] registerOnGdioChar(): GDIO characteristic registered
[ 40407][D][NukiBle.cpp:962] registerOnUsdioChar(): USDIO characteristic registered
[ 40514][D][NukiBle.cpp:674] pairStateMachine(): ##################### REQUEST REMOTE PUBLIC KEY #########################
Sending plain message: 01 00 03 00 27 a7 
[ 40524][D][NukiBle.cpp:901] sendPlainMessage(): Command identifier: 01, CRC: a727
[ 40662][D][NukiBle.cpp:987] notifyCallback():  Notify callback for characteristic: a92ae101-5501-11e4-916c-0800200c9a66 of length: 20
Received data: 03 00 17 ca 81 ff bc d1 6e d8 6b 6b f1 e6 e1 96 e2 d7 f6 b6 
[ 40675][E][NukiUtils.cpp:99] crcValid(): CRC CHECK FAILED!
[ 40707][D][NukiBle.cpp:987] notifyCallback():  Notify callback for characteristic: a92ae101-5501-11e4-916c-0800200c9a66 of length: 16
Received data: 51 22 a6 75 72 97 3c b7 af 3f b6 8c 5f 6b 33 e0 
[ 40721][E][NukiUtils.cpp:99] crcValid(): CRC CHECK FAILED!
[ 70556][W][NukiBle.cpp:814] pairStateMachine(): Pairing timeout
[ 70606][D][NukiBle.cpp:112] pairNuki(): pairing result 2
[ 72179][D][NukiBle.cpp:1179] onDisconnect(): BLE disconnected
[ 73086][D][NukiBle.cpp:254] onResult(): Found nuki in pairing state: Nuki_Opener_1B15B3A2 addr: NUKI BLE

technyon, 2023-04-16:
ok, thanks we'll have a look. Is your opener firmware up-to-date?

apfelflo89, 2023-04-16:
That might be the catch, I´m on an old firmware, since the support of my intercom got worse on nuki side and even dropped.

I'm on 1.3.1

technyon, 2023-04-16:
Why do they remove support for intercoms?

The latest version is 1.10.1, but you can't try since downgrading isn't possible. I'm not aware of changes to the BLE API though. There was a security update, but that wasn't a breaking change as far as I know.

apfelflo89, 2023-04-18:
Ok, I got a second (white) Nuki Opener for testing now...
What I can tell is that the delivered firmware was 1.5.3 and the pairing mode was not noticed at all,
after updating to the current version 1.10.1 the paring mode was recognized and the pairing process went through.
I prosume something is different with the paring process after some releases.

technyon, 2023-04-19:
That's good to know, I'll add it the readme. Maybe you could use another ESP to talk to your intercom, and wire the ESP to your opener. That way you could update and still use your unsupported intercom.

apfelflo89, 2023-04-21:
So I can confirm now: The error takes only place with the older firmwares. With 1.3.1 on my old Opener I had the CRC errors, right after the firmware update to 1.10.1 the error disappeared and the pairing went through with now error at all.
I guess this issue can be closed, I just checked the readme where you added the hint to the newest firmware already.