technyon/nuki_hub #176: Nuki Hub not pairing with new SL

bcutter, 2023-04-15:
Today I had to switch a SL 3.0 PRO.

Steps:

1. Nuki Hub: unpaired SL
2. Switched hardware
3. Configured new SL via Nuki app (everything is working fine)
4. Pair Nuki Hub with new SL 3.0

I'm stuck at step 4. No pairing possible strictly following https://github.com/technyon/nuki_hub#pairing

Done without success:
- restarted SL (battery removed for some time) several times
- gone through pairing process a dozen times
- restarted Nuki Hub several times

Nothing. At the moment it feels like 8.22 is unable to pair. No idea why Nuki Hub does not pair with the new SL. **HELP really appreciated** 🙏

bcutter, 2023-04-15:
Tried with Nuki Hub 8.18 and 8.8 now, made no difference.

Only things changed for the pairing scenario is:
- another SL 3.0 PRO (same type, different device)
- new SL firmware (now 3.5.12, no idea what was the latest stable out there when I paired the last time in February https://github.com/technyon/nuki_hub/issues/126#issuecomment-1432153266)

ESP is the same, Nuki Hub (firmware + config) is the same... what the heck.

**How to get some insights? Logs?
How can I check if the Nuki Hub REALLY starts to search for a SL in pairing mode and actually wants to pair with it?**

Not making any progress here, wasting one hour after another.

bcutter, 2023-04-15:
Latest update:

The very same NUKI Hub can pair almost immediately with the old SL. Tried several times. One time also with both - old and new - SL in pairing mode, powered on Nuki hub physically next to the new one - it only paired with the old SL.

I'm completely out of ideas. Remaining questions:
- Did Nuki hub REALLY unpair from the old SL?
- Is there some kind of "pairing cache" so Nuki Hub only pairs with the SL it was already paired in the past?

bcutter, 2023-04-15:
Another update:

ESP/Nuki hub 1: can only pair with old SL (the one it was paired before/first)
ESP/Nuki hub 2: can only pair with new SL (the one it was paired before/first)

I see some kind of pattern: **the one it was paired before/first**.

At the moment I put my bets on the unpairing is not done right in Nuki Hub.

I'm flashing Nuki Hub over ESP 2 to test this theory now.

technyon, 2023-04-16:
Give this binary a try.

[nuki_hub-8.23-pre-2.zip](https://github.com/technyon/nuki_hub/files/11241683/nuki_hub-8.23-pre-2.zip)

bcutter, 2023-04-16:
I'm good now. Hope I don't need to re-pair soon...

Did not test the linked pre-release.

technyon, 2023-04-16:
It would be helpful if you could test, since I can't really reproduce the issue. You don't need to unpair your though, you can test with one of your ESPs and enable "Register as app" before testing. That way your current ESPs don't get unpaired.

closed 2023-04-19 by technyon