technyon/nuki_hub #177: Feature request: Limit operations available through MQTT

mclei, 2023-04-17:
I would like to be able to limit the available operations through the MQTT. The reason is, that I only trust the native application to allow  "unlock" the smartlock. But I still want the full integration with Home Assistant using MQTT, so the state will update in HA or locking.

For example I want to lock the smartlock when some conditions are met, like leaving the home. But to unlock it, I need to use only my phone.

In the Web UI, there can be just checkboxes for every operation, like "unlock", "open", "lock", "ring to open", etc .... and I can enable/disable particular operation. Disabled operations will be just ignored when received through MQTT.

technyon, 2023-04-19:
What's your reason not to trust HA with controlling the lock?

Enabling/disabling lock actions altogether should be easy, basically it would be a read-only mode. Fine-grained control over every you can or can't do is more complicated. There's not only locking, but also making changes to the configuration, configuring keypad codes and so on.

mundschenk-at, 2023-04-19:
I can't speak for the OP, @technyon, but I am assuming the issue is remote access. You might want to be able to see whether the door is locked in an exposed HA instance, but not want an attacker to be able to use that access to open your front door. (Whether that's a likely attack scenario is worth another discussion of course.)

technyon, 2023-04-19:
I mean if that's issue, maybe it's better to secure the remote access ... but fair enough. Like I said, restricting all operations like locking, config changes and so on would be easy.

technyon, 2023-04-19:
I think I'd go for those three settings, as it's a lot easier to implement. And I can't really think of a use case where I only want to unlock but not to lock.

mclei, 2023-04-19:
> I think I'd go for those three settings, as it's a lot easier to implement. And I can't really think of a use case where I only want to unlock but not to lock.
Yes, its OK. Thanks.

technyon, 2023-05-06:
Hi,

you can give this binary a try. There's a drop down in the NUKI Configuration section to set the access level.

[nuki_hub-8.23-pre-5.zip](https://github.com/technyon/nuki_hub/files/11412020/nuki_hub-8.23-pre-5.zip)

technyon, 2023-05-07:
Implemented in version 8.23

closed 2023-05-07 by technyon