technyon/nuki_hub #178: Can't control two nuki locks separately in Home Assistant

Wpeter86, 2023-04-19:
Hello,
I have two Nuki Lock 3.0, each of which I have connected to an ESP32 flashed with nuki_hub (version 8.22). Both are also correctly detected individually in Home Assistant. However, both locks are always opened or closed, if i try to control one of them.

Do I have to change something in nuki_hub  to prevent this?

Thanks!

Wpeter86, 2023-04-19:
Found my mistake:
Had to rename "MQTT smart lock path" under "Basic Nuki Configuration". After that everything started working as excepted.

closed 2023-04-19 by Wpeter86

technyon, 2023-04-19:
Yes, maybe I'll add it to the readme, it's not that obvious you have to change it.

mundschenk-at, 2023-04-19:
Yes, with two locks you obviously need separate MQTT topics.